SpinalHDL/SpinalHDL #633: Enable single bus write per transaction to SPI master core.

Readon, 2022-03-06:
The existing SpiMasterCtrl can be operated for device which require SS in each transaction as below:
1. write 0x1100000X to data register to assert corresponding SS of X(SSid) slave.
2. write 0x000000XX or 0x010000XX to data register to do Read or Write on bus.
3. write 0x1000000X to dessert X slave.
It's too redundant when writing a program to operate on those devices.

Add flags to data register can help. Let the most significant bit represent if it is a compact operation.
For example, if write 0xYZ0000XX to data register where YZ is (0x80 + SSid) then it leads to assert SS before data operation and dessert SS after that. Then the software on operating the SPI can be simplified.

The implementation is also as simple as to extend the instruction with compact operation flag enabled into 3 commands shown above, so that the existing stable FSM can be remained.

Dolu1990, 2022-03-11:
> Then the software for operating the SPI can be simplified.

Why not abstracting that in a additional SPI driver function (soft) ?
I mean,  in general, i try to keep the hardware as simple as possible to save area, unless there is performances reasons.

Readon, 2022-03-11:
> > Then the software for operating the SPI can be simplified.
> 
> Why not abstracting that in a additional SPI driver function (soft) ? I mean, in general, i try to keep the hardware as simple as possible to save area, unless there is performances reasons.

That is also a solution. I have thought about it, the existing solution could not be easily implement this feature unless refactor it.

closed 2022-03-11 by Readon