SpinalHDL/SpinalHDL #638: `mergeRTLSource()` doesn't use `targetDirectory` in `SpinalConfig`

huaixv, 2022-03-10:
I use `SpinalConfig.targetDirectory` to specify where generated source files should be placed,
but if I use some `BlackBox`s in my design and use `mergeRTLSource` to merge them, the
generated blackbox file will be placed in project root instead of the `targetDirectory`.

I read the definition of `mergeRTLSource()` and find it doesn't use `targetDirectory` at all.
So is this the expected behavior?

A simple example here:
```scala
// MyComponent.scala
import spinal.core._

object MyConfig extends SpinalConfig(targetDirectory = "./build")

class MyBlackBox extends BlackBox { addRTLPath("MyBB.v") }
class MyComponent extends Component { val bb = new MyBlackBox() }

object MyVerilog {
  def main(args: Array[String]): Unit = SpinalVerilog(MyConfig)(new MyComponent).mergeRTLSource()
}
```
```verilog
// MyBB.v
module MyBlackBox(); endmodule
```
```shell
# tree
.
├── build
│   └── MyComponent.v
└── MyComponent_bb.v
```

Dolu1990, 2022-03-14:
Fixed ^^

closed 2022-03-14 by Dolu1990

huaixv, 2022-03-14:
Thank you very much. :+1:

commit a9ab7646 (2022-03-14): Fix #638 (mergeRTLSource now target the config targetdirectory)