SpinalHDL/SpinalHDL #639: the CI failed while "install cached tools" with error that ghdl is missing.

Readon, 2022-03-10:
I have tracked the log, find that GNAT installation failed while preparing ghdl. Here is the log.
[image]

Readon, 2022-03-11:
I tried to update the version number of tools which force recache the repos. 
It still report this problem even with [fix].(https://github.com/SpinalHDL/SpinalHDL/commit/4b19e7b2bb1984776713656ebbe02a405d52e33d)

Dolu1990, 2022-03-11:
Hmm i think this is another issue : 
/home/runner/.local/lib/python3.8/site-packages/cocotb/share/makefiles/simulators/Makefile.ghdl:51: *** "Unable to locate command >ghdl<".  Stop.

Because of 
2022-03-11T01:38:47.7417980Z Cloning into 'ghdl-build'...
2022-03-11T01:38:55.0281354Z HEAD is now at 2fb2384d Prepare release 1.0.0
2022-03-11T01:38:55.0516082Z Sorry, you need GNAT to build GHDL.  See the README

So seems we need to install packages before isntalling GHDL

Dolu1990, 2022-03-11:
Pushed a fix

Readon, 2022-03-11:
the cause of this error is missing of gnat.

 

 /home/runner/.local/lib/python3.8/site-packages/cocotb/share/makefiles/simulators/Makefile.ghdl:51: *** "Unable to locate command &gt;ghdl<".  Stop.

closed 2022-03-14 by Readon