SpinalHDL/SpinalHDL #641: Thread number calculation error BUG

dreamflyings, 2022-03-11:
## SpinalHDL Simulation Problem Description

When I simulation code with an Intel I9 12900H CPU(It is a CPU with 6 large cores + 8 small cores, 12+8=20 threads), I have the following problem

```
[Progress] Verilator compilation started
[info] Found cached verilator binaries

cpuDetails.size= 20 != sockets: 1 * coresPerSocket: 14 * threadsPerCore: 2
CpuInfo{socketId=0, coreId=0, threadId=0}
CpuInfo{socketId=0, coreId=0, threadId=1}
CpuInfo{socketId=0, coreId=4, threadId=0}
CpuInfo{socketId=0, coreId=4, threadId=1}
CpuInfo{socketId=0, coreId=8, threadId=0}
CpuInfo{socketId=0, coreId=8, threadId=1}
CpuInfo{socketId=0, coreId=12, threadId=0}
CpuInfo{socketId=0, coreId=12, threadId=1}
CpuInfo{socketId=0, coreId=16, threadId=0}
CpuInfo{socketId=0, coreId=16, threadId=1}
CpuInfo{socketId=0, coreId=20, threadId=0}
CpuInfo{socketId=0, coreId=20, threadId=1}
CpuInfo{socketId=0, coreId=24, threadId=0}
CpuInfo{socketId=0, coreId=25, threadId=0}
CpuInfo{socketId=0, coreId=26, threadId=0}
CpuInfo{socketId=0, coreId=27, threadId=0}
CpuInfo{socketId=0, coreId=28, threadId=0}
CpuInfo{socketId=0, coreId=29, threadId=0}
CpuInfo{socketId=0, coreId=30, threadId=0}
CpuInfo{socketId=0, coreId=31, threadId=0}

java.lang.AssertionError: cpuDetails.size= 20 != sockets: 1 * coresPerSocket: 14 * threadsPerCore: 2
CpuInfo{socketId=0, coreId=0, threadId=0}
CpuInfo{socketId=0, coreId=0, threadId=1}
CpuInfo{socketId=0, coreId=4, threadId=0}
CpuInfo{socketId=0, coreId=4, threadId=1}
CpuInfo{socketId=0, coreId=8, threadId=0}
CpuInfo{socketId=0, coreId=8, threadId=1}
CpuInfo{socketId=0, coreId=12, threadId=0}
CpuInfo{socketId=0, coreId=12, threadId=1}
CpuInfo{socketId=0, coreId=16, threadId=0}
CpuInfo{socketId=0, coreId=16, threadId=1}
CpuInfo{socketId=0, coreId=20, threadId=0}
CpuInfo{socketId=0, coreId=20, threadId=1}
CpuInfo{socketId=0, coreId=24, threadId=0}
CpuInfo{socketId=0, coreId=25, threadId=0}
CpuInfo{socketId=0, coreId=26, threadId=0}
CpuInfo{socketId=0, coreId=27, threadId=0}
CpuInfo{socketId=0, coreId=28, threadId=0}
CpuInfo{socketId=0, coreId=29, threadId=0}
CpuInfo{socketId=0, coreId=30, threadId=0}
CpuInfo{socketId=0, coreId=31, threadId=0}

	at net.openhft.affinity.impl.VanillaCpuLayout.<init>(VanillaCpuLayout.java:63)
	at net.openhft.affinity.impl.VanillaCpuLayout.fromCpuInfo(VanillaCpuLayout.java:142)
	at net.openhft.affinity.impl.VanillaCpuLayout.fromCpuInfo(VanillaCpuLayout.java:100)
	at net.openhft.affinity.impl.VanillaCpuLayout.fromCpuInfo(VanillaCpuLayout.java:95)
	at spinal.sim.VerilatorBackend$$anonfun$compileVerilator$2.apply$mcV$sp(VerilatorBackend.scala:633)
	at spinal.sim.VerilatorBackend.cacheSynchronized(VerilatorBackend.scala:74)
	at spinal.sim.VerilatorBackend.compileVerilator(VerilatorBackend.scala:600)
	at spinal.sim.VerilatorBackend.<init>(VerilatorBackend.scala:717)
	at spinal.core.sim.SpinalVerilatorBackend$.apply(SimBootstraps.scala:139)
	at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:701)
	at spinal.core.sim.SpinalSimConfig.compileCloned(SimBootstraps.scala:645)
	at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:622)
	at mylib.phigentai.ImageProcessTester$$anonfun$1.apply$mcV$sp(ImageProcessTester.scala:181)
	at mylib.phigentai.ImageProcessTester$$anonfun$1.apply(ImageProcessTester.scala:122)
	at mylib.phigentai.ImageProcessTester$$anonfun$1.apply(ImageProcessTester.scala:122)
	at org.scalatest.OutcomeOf$class.outcomeOf(OutcomeOf.scala:85)
	at org.scalatest.OutcomeOf$.outcomeOf(OutcomeOf.scala:104)
	at org.scalatest.Transformer.apply(Transformer.scala:22)
	at org.scalatest.Transformer.apply(Transformer.scala:20)
	at org.scalatest.funsuite.AnyFunSuiteLike$$anon$1.apply(AnyFunSuiteLike.scala:190)
	at org.scalatest.TestSuite$class.withFixture(TestSuite.scala:196)
	at org.scalatest.funsuite.AnyFunSuite.withFixture(AnyFunSuite.scala:1563)
	at org.scalatest.funsuite.AnyFunSuiteLike$class.invokeWithFixture$1(AnyFunSuiteLike.scala:187)
	at org.scalatest.funsuite.AnyFunSuiteLike$$anonfun$runTest$1.apply(AnyFunSuiteLike.scala:200)
	at org.scalatest.funsuite.AnyFunSuiteLike$$anonfun$runTest$1.apply(AnyFunSuiteLike.scala:200)
	at org.scalatest.SuperEngine.runTestImpl(Engine.scala:306)
	at org.scalatest.funsuite.AnyFunSuiteLike$class.runTest(AnyFunSuiteLike.scala:200)
	at org.scalatest.funsuite.AnyFunSuite.runTest(AnyFunSuite.scala:1563)
	at org.scalatest.funsuite.AnyFunSuiteLike$$anonfun$runTests$1.apply(AnyFunSuiteLike.scala:233)
	at org.scalatest.funsuite.AnyFunSuiteLike$$anonfun$runTests$1.apply(AnyFunSuiteLike.scala:233)
	at org.scalatest.SuperEngine$$anonfun$traverseSubNodes$1$1.apply(Engine.scala:413)
	at org.scalatest.SuperEngine$$anonfun$traverseSubNodes$1$1.apply(Engine.scala:401)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at org.scalatest.SuperEngine.traverseSubNodes$1(Engine.scala:401)
	at org.scalatest.SuperEngine.org$scalatest$SuperEngine$$runTestsInBranch(Engine.scala:396)
	at org.scalatest.SuperEngine.runTestsImpl(Engine.scala:475)
	at org.scalatest.funsuite.AnyFunSuiteLike$class.runTests(AnyFunSuiteLike.scala:233)
	at org.scalatest.funsuite.AnyFunSuite.runTests(AnyFunSuite.scala:1563)
	at org.scalatest.Suite$class.run(Suite.scala:1112)
	at org.scalatest.funsuite.AnyFunSuite.org$scalatest$funsuite$AnyFunSuiteLike$$super$run(AnyFunSuite.scala:1563)
	at org.scalatest.funsuite.AnyFunSuiteLike$$anonfun$run$1.apply(AnyFunSuiteLike.scala:237)
	at org.scalatest.funsuite.AnyFunSuiteLike$$anonfun$run$1.apply(AnyFunSuiteLike.scala:237)
	at org.scalatest.SuperEngine.runImpl(Engine.scala:535)
	at org.scalatest.funsuite.AnyFunSuiteLike$class.run(AnyFunSuiteLike.scala:237)
	at org.scalatest.funsuite.AnyFunSuite.run(AnyFunSuite.scala:1563)
	at org.scalatest.tools.SuiteRunner.run(SuiteRunner.scala:45)
	at org.scalatest.tools.Runner$$anonfun$doRunRunRunDaDoRunRun$1.apply(Runner.scala:1320)
	at org.scalatest.tools.Runner$$anonfun$doRunRunRunDaDoRunRun$1.apply(Runner.scala:1314)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at org.scalatest.tools.Runner$.doRunRunRunDaDoRunRun(Runner.scala:1314)
	at org.scalatest.tools.Runner$$anonfun$runOptionallyWithPassFailReporter$2.apply(Runner.scala:972)
	at org.scalatest.tools.Runner$$anonfun$runOptionallyWithPassFailReporter$2.apply(Runner.scala:971)
	at org.scalatest.tools.Runner$.withClassLoaderAndDispatchReporter(Runner.scala:1480)
	at org.scalatest.tools.Runner$.runOptionallyWithPassFailReporter(Runner.scala:971)
	at org.scalatest.tools.Runner$.run(Runner.scala:798)
	at org.scalatest.tools.Runner.run(Runner.scala)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.runScalaTest2or3(ScalaTestRunner.java:38)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.main(ScalaTestRunner.java:25)

Process finished with exit code 0

```
##Solution
The above problem can be solved if I make the following modifications. But this is a stopgap solution, and the solution is not perfect. Is there a better strategy?
grateful!

```
VerilatorBackend.scala
G633      val threadCount = if(isWindows || isMac) Runtime.getRuntime().availableProcessors() else 20//VanillaCpuLayout.fromCpuInfo().cpus()
```

Dolu1990, 2022-03-11:
Hi ^^

> Runtime.getRuntime().availableProcessors()

This isn't good, as that function return different things depending the load on the PC.

Can you try update the https://mvnrepository.com/artifact/net.openhft/affinity
Version ? 

Changing it here : 
https://github.com/SpinalHDL/SpinalHDL/blob/dev/build.sbt#L114
To let's say : 3.21ea83

(https://mvnrepository.com/artifact/net.openhft/affinity)

If then that's good, we can move dev to it ^^

dreamflyings, 2022-03-11:
> 

3.21ea83 is good~
bug fixed!
thanks!

closed 2022-03-11 by dreamflyings

Dolu1990, 2022-03-14:
merged ^^

commit a09612b8 (2022-03-14): Fix #641 net.openhft.affinity not working on some new CPU