SpinalHDL/SpinalHDL #643: In what cases can Spinal clone a bundle? and maybe a possible improvement

wmj0607, 2022-03-11:
Noticed that in some cases Spinal can't clone a bundle, wrap it with a `Flow` will fix it (then we can simply assign the valid signal with don't care and ignore it). but I can't get why this happens.

I wonder if we can make something similar(i.e. a Flow without valid) to provide another solution (in addition to the case class one). 
If we have a anonymous bundle who's parameter in it's closure, it's inconvenient to make it a case class (maybe there's way but i don't know?)

Some examples

```scala
case class nestedBundle (n:Int) extends Bundle{
  val raw = new Bundle{ //How to make this case class?
      val foo = UInt(n.bits)
    }
  val flow = Flow(new Bundle{
      val foo = UInt(n.bits)
    })
}

object cloneFoo{
  def raw(b:nestedBundle) = b.raw.clone()
  def flow(b:nestedBundle) = b.flow.clone()
}

object Test {
  def main(args: Array[String]) {SpinalConfig(mode = Verilog).generate{new Component{
    val bundle = new nestedBundle(4)
    cloneFoo.flow(bundle) //can clone
    cloneFoo.raw(bundle) //can't clone
  }}}
}
```

wswslzp, 2022-03-13:
Refer to #313 
`clone` function for anonymous `Bundle` is not allowed. `HardType` is a workaround. Redefine `cloneFoo.raw` as 
```scala
def raw(b:nestedBundle) = HardType(new Bundle {/*same as `raw` bundle*/})()
```

wmj0607, 2022-03-13:
fixed by define `nestedBundle.raw` as
```scala 
val raw= HardType(new Bundle{/* ... */})()
```

closed 2022-03-13 by wmj0607

Dolu1990, 2022-03-14:
ultimatly, maybe extending the SpinalHDL compiler plugin could cleanly solve the issue