SpinalHDL/SpinalHDL #646: `readmemb` forget to escape for path

kazutoiris, 2022-03-15:
I tried to simulate the script, but it will throw out `Unknown escape sequence: \U`.

```scala
val source = Source.fromFile("rom.bin","UTF-8")
```

In the Verilog file,

```verilog
$readmemb("C:\Users\Administrator\IdeaProjects\SpinalHDL\.\simWorkspace\MyTopLevel\rtl/MyTopLevel.v_toplevel_myRom_rom_1.bin",rom_1);
```

But truly it would be,

```verilog
$readmemb("C:/Users/Administrator/IdeaProjects/SpinalHDL/./simWorkspace/MyTopLevel/rtl/MyTopLevel.v_toplevel_myRom_rom_1.bin",rom_1);
```

This is obviously a problem caused by not escaping the path.

https://github.com/SpinalHDL/SpinalHDL/blob/b1c499841c99a03409bec081a52249e7f28a6955/core/src/main/scala/spinal/core/internals/ComponentEmitterVerilog.scala#L1138-L1155

Dolu1990, 2022-03-21:
Hi ^^

I'm not sure of the setup you are in, can you provide a simple self contained example ?

The issue may be also due to : 
https://github.com/SpinalHDL/SpinalHDL/blob/d80e5f987323bbb3d202a2e241e8e025651322d3/core/src/main/scala/spinal/core/sim/SimBootstraps.scala#L750

kazutoiris, 2022-03-21:
I created a new repository for the full project [refactored-tribble](https://github.com/kazutoiris/refactored-tribble).

[MyTopLevel.v](https://github.com/kazutoiris/refactored-tribble/blob/main/MyTopLevel.v) is generated by `SpinalVerilog`, [simWorkspace/MyTopLevel/MyTopLevel.v](https://github.com/kazutoiris/refactored-tribble/blob/main/simWorkspace/MyTopLevel/rtl/MyTopLevel.v) & [tmp/job_1/MyTopLevel.v](https://github.com/kazutoiris/refactored-tribble/blob/main/tmp/job_1/MyTopLevel.v) are generated by `SimConfig.withWave.doSim`.

In `MyTopLevel.v`

https://github.com/kazutoiris/refactored-tribble/blob/c3c3800af2af054d16ebebbfe9c5b0c8fbe5a5a6/MyTopLevel.v#L43

But in `simWorkspace/MyTopLevel/MyTopLevel.v`,

https://github.com/kazutoiris/refactored-tribble/blob/c3c3800af2af054d16ebebbfe9c5b0c8fbe5a5a6/simWorkspace/MyTopLevel/rtl/MyTopLevel.v#L43

The full verilator output is

```
%Error: C:/Users/Microsoft/IdeaProjects/SpinalMIPS/tmp/job_1/MyTopLevel.v:43: Unknown escape sequence: \U
%Error: Exiting due to 1 error(s)
```

Dolu1990, 2022-03-23:
Ahh, seems like https://github.com/kazutoiris/refactored-tribble/tree/main/src/main/scala is empty ? i mean can't go into the folder.

kazutoiris, 2022-03-23:
Sorry, I forget to migrate the sub project. Now it's can be accessed.

Dolu1990, 2022-03-25:
Thanks ^^

I just tested, this is fixed in the dev branch, so you use it via : 
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/About%20SpinalHDL/faq.html#how-to-use-an-unreleased-version-of-spinalhdl-but-committed-on-git

closed 2022-03-25 by Dolu1990