SpinalHDL/SpinalHDL #647: need 'XOR' 'OR' for mask type

ArcheyChen, 2022-03-15:
for now ,mask type is only use for comparison,like `someUInt === M"--1-"`
But I think some times we want to use 'OR' 'XOR'
like 
`someUInt := A | M'--1-'`(set A(1) to 1)

Dolu1990, 2022-03-16:
Hmm, what's about : 
`someUInt := A | B'0010'`  instead ?
I mean, it can make sense to use MaskedLiteral as you showed, but doesn't it make more sense to use the native data type ?

ArcheyChen, 2022-03-16:
> Hmm, what's about : `someUInt := A | B'0010'` instead ? I mean, it can make sense to use MaskedLiteral as you showed, but doesn't it make more sense to use the native data type ?

well in most case, we do use something like `A | B'0010'`, but I'm designing a bus , and I want to write something like:
```scala
val isWrite = M'--1-'// the 2nd bit is write
when(param === isWrite){}// decode the param
///////////////////
val param = someVal
val newParam = param | isWrite // add info to param
```

of course, I can make `isWrite` as a UInt, and using something like`when((param & isWrite).orR){}`, but it's not that handy

tanhongze, 2022-04-16:
I think maybe we could add some methods to MaskedLiteral
```
val mask = M"-01-"
val sensitive = masked.sensitive // B"0110"
val value = masked.fillTo(false) // B"0010", or shorted by masked.filled
```
So, we can make it like:
```
val isWrite = M'--1-'// the 2nd bit is write
when(param === isWrite){}// decode the param
///////////////////
val param = someVal
val newParam = param | isWrite.filled
```

Dolu1990, 2022-04-18:
I think, having something in the kind masked.filleto(x) would be good, yes.

We could directly have   
- masked.asUInt(x)  //x being the filling, default being False
- masked.asBits(x)  //x being the filling, default being False

Dolu1990, 2022-04-18:
Fix pushed in dev via the cast methodes :)

Sorry for the delay, i losed track of that issue. You can ping me when you see it happen ^^

commit 6e850601 (2022-04-18): fix #647 Add MaskedLiteral casts to Bits/UInt/SInt
closed 2022-04-18 by Dolu1990