SpinalHDL/SpinalHDL #648: SpinalEnum support print

pwang7, 2022-03-15:
Currently, SpinalEnum does not support print in simulation, that:
```
println(${some SpinalEnum val}
```
will print it as `null`.

Dolu1990, 2022-03-16:
I just tested it seems to work on my side.

Can you provide a minmal self contained code for me to run ?

pwang7, 2022-03-18:
I wrote a test piece, and it works.
Thanks!

```Scala
import spinal.core._
import spinal.core.sim._
import spinal.lib._

class SpinalEnumTest extends AnyFunSuite {
  object WorkReqOpCode extends SpinalEnum {
    val RDMA_WRITE, RDMA_WRITE_WITH_IMM, SEND, SEND_WITH_IMM, RDMA_READ,
        ATOMIC_CMP_AND_SWP, ATOMIC_FETCH_AND_ADD, LOCAL_INV, BIND_MW,
        SEND_WITH_INV, TSO, DRIVER1 = newElement()

    defaultEncoding = SpinalEnumEncoding("opt")(
      RDMA_WRITE -> 0,
      RDMA_WRITE_WITH_IMM -> 1,
      SEND -> 2,
      SEND_WITH_IMM -> 3,
      RDMA_READ -> 4,
      ATOMIC_CMP_AND_SWP -> 5,
      ATOMIC_FETCH_AND_ADD -> 6,
      LOCAL_INV -> 7,
      BIND_MW -> 8,
      SEND_WITH_INV -> 9,
      TSO -> 10,
      DRIVER1 -> 11
    )
  }

  class SpinalEnumExample extends Component {
    val io = new Bundle {
      val workReqOpCode = in(WorkReqOpCode())
    }
    val opcode = RegNext(io.workReqOpCode) init (WorkReqOpCode.TSO)

    report(L"${REPORT_TIME} time: in report ${opcode}")
  }

  val simCfg = SimConfig.allOptimisation.withWave
    .compile(new SpinalEnumExample)

  test("FragmentStreamForkQueryJoinResp no query test") {
    simCfg.doSim { dut =>
      dut.clockDomain.forkStimulus(10)

      val workReqOpCode = WorkReqOpCode.RDMA_READ
      dut.io.workReqOpCode #= workReqOpCode

      dut.clockDomain.waitSampling(3)
      println(f"${simTime()} time: in sim ${workReqOpCode}")
    }
  }
}

```

Dolu1990, 2022-03-21:
I got : 

```
[Progress] Start SpinalEnumExample test simulation with seed 1933650510
NOTE                    0 time: in report TSO                 
NOTE                    0 time: in report RDMA_READ           
190 time: in sim RDMA_READ
[Done] Simulation done in 9.018 ms
```

Which seems good right ?

Which version of SpinalHDL do you have ?

pwang7, 2022-03-21:
> I got :
>
> ```
> [Progress] Start SpinalEnumExample test simulation with seed 1933650510
> NOTE                    0 time: in report TSO
> NOTE                    0 time: in report [...]
Yes.
> 
> Which version of SpinalHDL do you have ?
This test script is of 1.6.4, my project still use 1.6.2, let me try upgrade first.

numero-744, 2022-10-14:
Hello @pwang7 did the upgrade fix your issue?

pwang7, 2022-10-15:
> Hello @pwang7 did the upgrade fix your issue?

Yes, this PR can be closed.

closed 2022-10-15 by pwang7