SpinalHDL/SpinalHDL #652: The order of asyncReset, syncReset and clockEnable

NomadShen, 2022-03-17:
In `ComponentEmitterVerilog.scala`, the `emitClockedProcess` deal with asyncReset and clockEnable before softReset and syncReset. It can cause that the syncReset can't work since it is blocked by clockEnable.

Maybe an opinion could be added to config the priority of reset and clock enable.
I think an option to decide the order of them can be added in the spinalConfig. It is because those signals are relatively fixed in the `emitClockedProcess`.

Dolu1990, 2022-03-21:
hmm, i think one of the concern i had was to have things inferring well in FPGA flops.
But i agree that from a behavioral standpoint it make sense to have it parametrable. Overall, i was thinking by the past to have a less constrained ClockDomain class, in which you can define multiple reset sources / clock enable with different priorities.

I know that's a workaround, but what about you force the clockenable high when you want to do a soft reset ?

NomadShen, 2022-03-21:
这是来自QQ邮箱的假期自动回复邮件。您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

NomadShen, 2022-03-21:
> one of the concern i had was to have things inferring well in FPGA flops.

Unlike FPGA, in ASIC design, tools like the Design Compiler can handle the integrate clock gating cells.

> Overall, i was thinking by the past to have a less constrained ClockDomain class, in which you can define multiple reset sources / clock enable with different priorities.

The idea is quiet cool!

> what about you force the clockenable high when you want to do a soft reset ?

I rewrite the clockEnable logics as the way I used to arrange them in Verilog. The classic method always works :)
It's a pity the powerful clock domain feature doesn't work so well here. But it's all right.

Furthermore, what about meationing the priorities of these signal in the Spinal Doc? An explanation and a small example can be helpful to those who encounter a similar bug.

Dolu1990, 2022-03-23:
> It's a pity the powerful clock domain feature doesn't work so well here. But it's all right.

That's one of the challenge of designing SpinalHDL itself, aiming far enough from the beggining ^^
For clock domains, it wasn't far enough XD

> Furthermore, what about meationing the priorities of these signal in the Spinal Doc? An explanation and a small example can be helpful to those who encounter a similar bug.

Sure, do you feel you can do a PR on :
https://github.com/SpinalHDL/SpinalDoc-RTD/blob/004ed6243f19089af61098abe5104ba8388f0037/source/SpinalHDL/Structuring/clock_domain.rst 
?

NomadShen, 2022-03-23:
OK. I will check the code and write a PR :)

closed 2022-03-23 by NomadShen