SpinalHDL/SpinalHDL #653: how to check a signal is unload?

dxsm, 2022-03-18:
Hi, is there a method to check a signal is unload? I thought `dlcIsEmpty` can be check unload, but I‘m Wrong...
```scala
val aa = B"3'd0"
val bb = aa
println(aa.isUnload())   //false
println(bb.isUnload())  //true
```
As the example show, how to check a signal is unload?

Dolu1990, 2022-03-21:
Hi,

I'm not sure to understand what you mean by unload ?

Do you mean unassigned ?

From a SpinalHDL standpoint, both aa and bb of your example are exactly the same

dxsm, 2022-03-22:
oh yeah, aa and bb is the same. As "a = b & c", b and c load to a, if a is not assigned to other nodes, we say a is unload. So how to check a node is unload?

Dolu1990, 2022-03-22:
Ahh, so, currently you can easily check if a node is assigned by something, but checking if a node is used to assign something require searching into the SpinalHDL graph.

There is currently no utilities which implement it. but it would not be very complicated to do.

Mostly : 

```scala
    y.component.dslBody.walkStatements{e =>
      e.walkDrivingExpressions{
        case `y` => {
          println("hit")
        }
        case _ =>
      }
    }
```

Dolu1990, 2022-03-22:
would also need to manage more walking if the signal is a io

numero-744, 2022-10-13:
I think the issue can be closed

closed 2022-10-17 by Dolu1990