SpinalHDL/SpinalHDL #654: How to update spinal to the latest version for existing projects

Aaronwangyilun, 2022-03-19:
I am a beginner, I would like to ask how to update spinal to the latest version for existing projects.

tanhongze, 2022-03-21:
If you starts your project from `SpinalTemplateSbt`, find `build.sbt`
The version of spinal is specified by a variable in that file
e.g. ```val spinalVersion = "1.6.4"```
To update spinal to the latest version, simply change it to the lastest

Aaronwangyilun, 2022-03-25:
Thanks a lot!~now I am using IDEA, can I only regenerate the new project with the new version of build.sbt and include the code in it?

> If you starts your project from `SpinalTemplateSbt`, find `build.sbt` The version of spinal is specified by a variable in that file e.g. `val spinalVersion = "1.6.4"` To update spinal to the latest version, simply change it to the lastest

Aaronwangyilun, 2022-03-25:
> If you starts your project from `SpinalTemplateSbt`, find `build.sbt` The version of spinal is specified by a variable in that file e.g. `val spinalVersion = "1.6.4"` To update spinal to the latest version, simply change it to the lastest

Thank you very much! I modified the build.sbt file and updated it, and I succeeded

Dolu1990, 2022-03-25:
> Thanks a lot!~now I am using IDEA, can I only regenerate the new project with the new version of build.sbt and include the code in it?

So normaly, you only need to do a refresh: 
view/tool windows/sbt and then =>
[image]

closed 2022-10-14 by numero-744