SpinalHDL/SpinalHDL #655: If the support of falling edges' actions during simulation have critical shortage?

Readon, 2022-03-19:
There are cases where there are two separately monitored streams whose payloads are functionally coupled and must be checked during simulation, such as the address and data of some buses or the inputs and outputs of combinational logic.
In the case of two streams fired simultaneously, the order of execution of the monitor callback functions cannot be determined because the stream monitors are parallelized simultaneously, resulting in check functions that cannot be written independently.

Therefore, the strategy that exists in Axi4Agent is to push the check function and data in the corresponding queue, and then call the same function in both callbacks to check if both queues are non-empty, and execute the check function to check the data only when the condition is met.

Another strategy is to allow StreamMonitor to not only trigger on the rising edge but also to add callbacks to the falling edge, and then have the preparation of the check data trigger on the falling edge and execute the check function on the rising edge.
This should be much easier than the existing approach, just having the preparation and check executed at different times corresponding to the edge state.

However, there are drawbacks, as I know that this is not friendly for post-synthesis simulation if the design has to be fully synchronized.
Are there any other shortcomings to be aware of?
Or how should the existing strategy be simplified?

Readon, 2022-03-21:
> 

How about add a QueuesScoreboard which accept several queues and only execute the check function when all queues are nonEmpty, with 1 unit of time delay?

Readon, 2022-08-31:
How about add a parameter "delay" to StreamMonitor so that the Monitor can execute callbacks in specified delayed time.
Which can help to work around on the problem of two streams fire at the same time, where the execution flow is random.

Readon, 2022-08-31:
A good solution is to use delayed function in StreamMonitor function, as below.
``` scala
StreamMonitor(stream1) { p => zzzz.enqueue(p) }

StreamMonitor(stream2) { p => {
    val xxx = p.check  // here is prepare for check.
    delayed(1) {
        assert(xxx === zzzz.dequeue()) // here is the real check.
    }
  }
}
```

Dolu1990, 2022-09-02:
> A good solution is to use delayed function in StreamMonitor function, as below.

Sometimes i defined one queue for each StreamMonitor in which a push stuff. And after each push i call a check function which will check if both queue have something.

Readon, 2022-09-02:
> > A good solution is to use delayed function in StreamMonitor function, as below.
> 
> Sometimes i defined one queue for each StreamMonitor in which a push stuff. And after each push i call a check function which will check if both queue have something.

It's the current solution of Axi4XXXXMonitor. However, it seems not that straight forward, and hard to debug. 
For testing, the logic should be as simple as possible to avoid introducing more bugs.