SpinalHDL/SpinalHDL #658: Support `assignUnassignedByName` for input/output?

tanhongze, 2022-03-21:
I found that input signals would be assigned by `assignUnassignedByName` currently, which causes `HIERARCHY VIOLATION`.
To support `assignUnassignedByName` for input/output, more cases should be considered.
It should also see whether signal is input of current model or output of sub-module.
So, should we update `hasDataAssignment` or add a new method?

Dolu1990, 2022-03-23:
Hi,

Can you provide a self contained example of the user code that you-would-like / break  ?

Thanks ^^

tanhongze, 2022-03-23:
For example, 
```
class FooBundle extends Bundle{
  val cond0 = in  Bool()
  val cond1 = in  Bool()
  val flag  = out Bool()
  val state = out UInt(8 bits)
}

class MySubLevel extends Component{
  val io = new FooBundle
  io.flag  := False
  io.state := U(0)
}

class MyTopLevel extends Component{
  val io = new FooBundle
  val sub = new MySubLevel

  // OK
  sub.io.cond0 := False
  sub.io.assignUnassignedByName(io)

  // HIERARCHY VIOLATION
  io.flag := False
  io.assignUnassignedByName(sub.io)

}
```

Dolu1990, 2022-03-24:
Ahhh make sense.

Dolu1990, 2022-03-24:
Should be good now, let's me know if you have any issue ^^

commit 307fb8b8 (2022-03-24): Add MultiData.zipByName(MultiData)
Fix #658