SpinalHDL/SpinalHDL #659: Fix `AxiMemorySim`?

tanhongze, 2022-03-21:
I found that `spinal.lib.bus.amba4.axi.sim.AxiMemorySim` does not obey AXI4 specification
There are a few sub-issues
1. Cannot correctly reply `rid` (not driven now)
2. read/write transactions is not correct when address is not aligned
e.g. 
We have `32'h3210` at address 0 and `32'h7654` at address 4.
The bus should reply `32'h3210` (generally `32'h--1-`) when address 1 is required and `arsize` is set to 0, instead of `32'h4321`

tanhongze, 2022-03-26:
Could I fix `AxiMemorySim`? Is there any requirement for PR?

c-thaler, 2022-03-26:
Your fix will be highly appreciated :D

Dolu1990, 2022-03-28:
> Is there any requirement for PR?

I would say, the requirement is to fix more than it break ^^

And to not break backward compatibility.

commit 3dc0bfe7 (2022-04-15): Fix #659 AxiMemorySim for AXI4 specification

1. support narrow transfer

2. support burst and write strobes

3. support id replying (arid->rid,awid->bid)

4. support read and write response

5. check burst does not cross 4KB boundary
commit 99f9a144 (2022-04-15): Merge pull request #672 from tanhongze/dev

Fix #659 AxiMemorySim for AXI4 specification
closed 2022-04-15 by Dolu1990