SpinalHDL/SpinalHDL #667: romReuse bug

jijingg, 2022-03-26:
**Bug1:**
I have 7 same rom instance ,  but now I got  tow mem_rom_wrap with two different path after this commit https://github.com/SpinalHDL/SpinalHDL/commit/8ea711515c4fc0cd31db798829a3216f5dd43737#r69580527
the `tmp/job_1`.  appear two time in second module ;

**Question :**
my simworkspace   is. `simWorkspace/fft4096Top4L4D/`
However, the rom.bin always genterate to  `tmp/job_x` not `simWorkspace/fft4096Top4L4D` during simulation   .
Is there any special reason for this？ @Dolu1990 

```verilog 
12062 //mem_rom_1025x32_fft4096_wrap_6 replaced by mem_rom_1025x32_fft4096_wrap_1
12064 //mem_rom_1025x32_fft4096_wrap_5 replaced by mem_rom_1025x32_fft4096_wrap_1
12066 //mem_rom_1025x32_fft4096_wrap_4 replaced by mem_rom_1025x32_fft4096_wrap_1
12068 //mem_rom_1025x32_fft4096_wrap_3 replaced by mem_rom_1025x32_fft4096_wrap_1
12070 //mem_rom_1025x32_fft4096_wrap_2 replaced by mem_rom_1025x32_fft4096_wrap_1
12071
12072 module mem_rom_1025x32_fft4096_wrap_1 (
...
12086   initial begin
12087     $readmemb("/Users/jijing/nutstore/gitlab/docker/lab/Maclab/tmp/job_1/tmp/job_1/fft4096Top4L4D.v_toplevel_uMain_uTwiddleGroup_uWRom_roms_0_rom_7_mem.bin",mem);
12088   end
...
12097 endmodule
12098 
12099 module mem_rom_1025x32_fft4096_wrap (
....
12113   initial begin
12114     $readmemb("/Users/jijing/nutstore/gitlab/docker/lab/Maclab/tmp/job_1/fft4096Top4L4D.v_toplevel_uMain_uTwiddleGroup_uWRom_roms_0_rom_7_mem.bin",mem);
12115   end
...
12124 endmodule        
```

Dolu1990, 2022-03-26:
Ahhhh got it, can reproduce it via : 

```scala
  SimConfig.withConfig(SpinalConfig(romReuse = true)).withFstWave.compile(new Component {
    val mem = Mem.fill(256)(UInt(16 bits)).initBigInt((0 to 255).map(BigInt(_)))
    out(mem.readAsync(in(mem.addressType)))

    val mem2 = Mem.fill(256)(UInt(16 bits)).initBigInt((0 to 255).map(BigInt(_)))
    out(mem2.readAsync(in(mem.addressType)))
  }).doSim{ dut =>
    sleep(10)
  }
  ``` 
  
I just fixed it.

> Is there any special reason for this

I don't realy remember, i would say it is mostly due to the fact that spinal.sim is a standalone module, while spinal.core.sim is on the top of it.

jijingg, 2022-03-27:
> I just fixed it.

thanks, it's work now .

> I don't realy remember, i would say it is mostly due to the fact that spinal.sim is a standalone module, while spinal.core.sim is on the top of it.

I saw here  force replace  targetDirectory with `tmp/job_x`，is it necessary?
https://github.com/SpinalHDL/SpinalHDL/blob/0948de5a9be75d843f482fc9a9ad072003aac4df/core/src/main/scala/spinal/core/sim/SimBootstraps.scala#L737

Dolu1990, 2022-03-28:
> I saw here force replace targetDirectory with tmp/job_x，is it necessary?

For now yes, mostly because spinal.core is decoupled from spinal.core.sim, and the second is using the name of the elaborated toplevel as workspace name.

closed 2022-03-30 by jijingg