SpinalHDL/SpinalHDL #669: VCS Simulation Error

HC153Yuanyuan, 2022-03-28:
I'm trying to use VCS simulation and have some problems. The SpinalHDL branch I used is dev@0948de5a9, and the test project is SpinalTemplateSbt. VCS log as following:

Command: vcs -full64 -quiet -timescale=1ns/1ps -debug_access+all -debug_acc+pp+dmptf \
-debug_region=+cell+encrypt -l vcs.log +vpi +vcs+initreg+random -load /home/zyy/workspace/SpinalTemplateSbt/./simWorkspace/.pluginsCachePath/vpi_vcs.so:entry_point_cb \
-o MyTopLevel -P /home/zyy/ssd/software/verdi/Verdi_O-2018.09-SP2/share/PLI/VCS/LINUX64/novas.tab \
/home/zyy/ssd/software/verdi/Verdi_O-2018.09-SP2/share/PLI/VCS/LINUX64/pli.a -kdb \
MyTopLevel __simulation_def

Warning-[DBGACC_REG] Unrecognized '-debug_region' option
  Option '' is not a valid '-debug_region' option.
  Please recompile using the '-debug_access<+options>' switch and incremental 
  options as required.  Recommended options are '-debug_access' for 
  post-process debug, '-debug_access+class' for testbench debug, and 
  '-debug_access+all' for all debug capabilities.  Refer the VCS user guide 
  for more granular options for debug control under the switch '-debug_access'
  and refer to '-debug_region' for region control.

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_register_cb' is not supported in the compiler.

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_chk_error' is not supported in the compiler.

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_chk_error' is not supported in the compiler.

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_register_cb' is not supported in the compiler.

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_chk_error' is not supported in the compiler.

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_chk_error' is not supported in the compiler.

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_register_cb' is not supported in the compiler.

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_chk_error' is not supported in the compiler.

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_chk_error' is not supported in the compiler.

Error-[SFCOR] Source file cannot be opened
  Source file "MyTopLevel" cannot be opened for reading due to 'No such file 
  or directory'.
  Please fix above issue and compile again.
  
  

10 warnings
1 error

wswslzp, 2022-03-28:
Could you show me your `vlogan.log` or `vhdlan.log`  as well?

HC153Yuanyuan, 2022-03-28:
> Could you show me your `vlogan.log` or `vhdlan.log` as well?
FYI:

Warning-[MXIR-W] VCS-MX build is required
  Please make sure that vlogan is from the intended build.

Warning-[LINX_KRNL] Unsupported Linux kernel
  Linux kernel '5.4.0-105-generic' is not supported.
  Supported versions are 2.4* or 2.6*.

Command: vlogan -nc -full64 -sverilog +v2k -ntb -debug_access+all -timescale=1ns/1ps \
-l vlogan.log -kdb /home/zyy/workspace/SpinalTemplateSbt/tmp/job_1/MyTopLevel.v ./rtl/__simulation_def.v \

Parsing design file '/home/zyy/workspace/SpinalTemplateSbt/tmp/job_1/MyTopLevel.v'
Parsing design file './rtl/__simulation_def.v'

wswslzp, 2022-03-28:
Could you provide the test case you used? I cannot reproduce your problem.

HC153Yuanyuan, 2022-03-28:
> Could you provide the test case you used? I cannot reproduce your problem.

my testcase is https://github.com/SpinalHDL/SpinalTemplateSbt , and the MyTopLevelSim.scala as following:

```
package mylib

import spinal.core._
import spinal.sim.vpi._
import spinal.sim._
import spinal.core.sim._
import scala.util.Random

//MyTopLevel's testbench
object MyTopLevelSim {
  def main(args: Array[String]) {
    val flags = VCSFlags(
        compileFlags = List("-kdb"),
        elaborateFlags = List("-kdb")
    )

    SimConfig
      .withVCS(flags)
      .withFSDBWave
      .workspacePath("tb")
      .doSim(new MyTopLevel){dut =>
      //Fork a process to generate the reset and the clock on the dut
      dut.clockDomain.forkStimulus(period = 10)

      var modelState = 0
      for(idx <- 0 to 99){
        //Drive the dut inputs with random values
        dut.io.cond0 #= Random.nextBoolean()
        dut.io.cond1 #= Random.nextBoolean()

        //Wait a rising edge on the clock
        dut.clockDomain.waitRisingEdge()

        //Check that the dut values match with the reference model ones
        val modelFlag = modelState == 0 || dut.io.cond1.toBoolean
        assert(dut.io.state.toInt == modelState)
        assert(dut.io.flag.toBoolean == modelFlag)

        //Update the reference model value
        if(dut.io.cond0.toBoolean) {
          modelState = (modelState + 1) & 0xFF
        }
      }
    }
  }
}
```

HC153Yuanyuan, 2022-03-28:
the vcs version: O-2018.09-SP2
the verdi version: Verdi_O-2018.09-SP2

C1oud555, 2022-03-30:
I come across this issue Too!

HC153Yuanyuan, 2022-03-30:
Hi all, I fixed it by reinstalling my VCS from another installation package. Maybe there is something function limitation with my original VCS. I have tried two versions which works well.
1. vcs script version : O-2018.09 Compiler version=VCS-MX O-2018.09-SP2_Full64
2. vcs script version : L-2016.06 Compiler version = VCS-MX L-2016.06_Full64

closed 2022-03-30 by HC153Yuanyuan