SpinalHDL/SpinalHDL #674: The Mem write enable function

liqx-github, 2022-04-03:
[image]

Hello! I'm having problems using Mem's write Enable function. As shown in the picture, The condition pickMem === U(i) will not work when I place io.memWr in the enable parameter of Mem, What kind of consideration is that?

Dolu1990, 2022-04-04:
Hi,

So the Mem API is made in a way where, either you don't specify the function enable argument, and you let the conditional scope do all the work, either you specify the function enable argeuement, and it will be the only thing considered.

So it is the expected behaviour, then maybe it isn't the best one.

Does it make sense ?

liqx-github, 2022-04-04:
I got it. Thank you very much!

closed 2022-04-04 by liqx-github