SpinalHDL/SpinalHDL #676: VCS simulation issue When a blackbox without reset

C1oud555, 2022-04-06:
When I try to simulate a blackbox ram without. It occurs `java.util.NoSuchElementException: key not found: (reset :  Bool)`
Just the document's BlackBox example with `SimConfig.withVCS.compile....`

Dolu1990, 2022-04-07:
Do you have a simple code example to reproduce the issue ?

C1oud555, 2022-04-09:
```
class nSram(addrWidth: Int, dataWidth: Int) extends BlackBox {
  val io = new Bundle {
    val clk   = in Bool ()
    val ramIO = slave(sramIO(addrWidth, dataWidth))
  }
  addGeneric("DW", dataWidth)
  addGeneric("AW", dataWidth)

  noIoPrefix()
  mapCurrentClockDomain(clock = io.clk)
  addRTLPath("./rtl/nSram.v")

  private def renameIO(): Unit = {
    io.flatten.foreach(implicit bt => {
      rename("ramIO_", "")
    })
  }
  // Execute the function renameIO after the creation of the component
  addPrePopTask(() => renameIO())
}
```
and the simulation
```
object sramTest extends App {
  val compiled = SimConfig.compile(new sramSim)

  compiled.doSim { dut =>
    dut.clockDomain.forkStimulus(10)
    dut.clockDomain.waitSampling(20)
  }
}
```
I know there's no reset signal in the blackbox. But It's just what the documents did.

numero-744, 2022-10-23:
I cannot reproduce the example. What is `sramIO`?

Dolu1990, 2022-10-24:
Ahhh damned, sorry i losed track of that issue :/

Dolu1990, 2022-10-24:
Ahhhh, so, and your toplevel never use the clockdomain for himself right ? 
and so mapCurrentClockDomain as you used it never interrect with the reset either.

So yes i can see how that crash happened.

In this kind of case, one workaround is to manualy use the clockdomain in the toplevel : 

```scala
new sramSim{RegNext(False)}
```

numero-744, 2022-10-24:
Doesn't it generate a reg? I usually use `Reg(Bits(0 bits))` so that there are no reg in the generated code and no pruned wire too.

Dolu1990, 2022-10-24:
> Doesn't it generate a reg?

Yes

> I usually use Reg(Bits(0 bits)) so that there are no reg in the generated code and no pruned wire too.

Ahhhhh lolololol   <3 XD

Dolu1990, 2022-10-25:
@C1oud555 I pushed a fix in dev, now it should be fine.

closed 2022-11-21 by numero-744