SpinalHDL/SpinalHDL #677: How about add a regroup function for BitVector?

Readon, 2022-04-06:
sometimes, a regroup on bits is required, such as D[31:0] ==> {D[7:0], D[15:8], D[23:16], D[31:24]}.
how about add a regroup function for BitVector, so that the usage can be as below.
``` scala
     val a = Bits(32 bits)
     a.regroup(8 bits) { (i, total) =>
         total - 1 - i
     }
     // the implementation could be 
     def regroup(sliceWidth bits)(indexMap: (Int, Int) => Int) {
          val inVec = this.subdivideIn(sliceWidth)
          val outVec = cloneOf(inVec)
          for(i <- inVec.indexes) {
              outVec(indexMap(i)) := inVec(i)
          }
         val out = cloneOf(this)
         out := outVec.asBits.asIntance[type(this)]   //this might require correction.
     }
```

tanhongze, 2022-04-06:
`D[31:0] ==> {D[7:0], D[15:8], D[23:16], D[31:24]}` is already implemented as `EndiannessSwap`

Readon, 2022-04-06:
> `D[31:0] ==> {D[7:0], D[15:8], D[23:16], D[31:24]}` is already implemented as `EndiannessSwap`

em, it's just an example, it could be more flexible.

Dolu1990, 2022-04-07:
Hi,

hmmm maybe this could be implemened by adding / composing smaller primitives ?

a.subdivideIn(8 bits).shuffle(x => y).asBits ?

Having both subdibide and shuffle inside a single function seems a lot ?

Readon, 2022-04-07:
> e inside a single function seem

yes, that would be better to keep things simple. 
Then there should be a shuffle method of Vec, which also leave the type problem to user when output type is not Bits.

PR #681 (2022-04-13): shuffle function for Vec.
fix #677
closed 2022-04-13 by Dolu1990