SpinalHDL/SpinalHDL #686: About adding a constructor for Axi4SlaveFactory?

NomadShen, 2022-04-11:
The implementations of the BusSlaveFactory on Apb3 or AvalonMM buses have the constructors. For example:
```scala
object Apb3SlaveFactory {
  def apply(bus: Apb3, selId: Int = 0, dontCareReadData : Boolean = false) = new Apb3SlaveFactory(bus, selId, dontCareReadData)
}
```
However, the BusSlaveFactory of Axi4 and Axi4lite don't have one, which makes some incoherence in the lib. How about fixing it by a PR?

Besides, I found many codes in the lib are not well documented, such as those bus implementation. Maybe we can add some effort to making the SpinalDoc better :)

NomadShen, 2022-04-12:
I found that BusSlaveFactory and BusInterface(regif) have similar functionalities. Mentioning it in the doc could be helpful to users.
Though both of them are amazing, is there any plan to merge them for simplification? For instance, use BusSlaveFactory as a backend of register interface.
Just a small suggestion, I'm not familiar enough with lib.bus :)

Dolu1990, 2022-04-13:
> is there any plan to merge them for simplification ?

So currently there is no real plan for it. Of course it would be better, issue is man power and will :)

> For instance, use BusSlaveFactory as a backend of register interface.

Yes i think that would make sense.

closed 2022-04-18 by NomadShen