SpinalHDL/SpinalHDL #694: whenCompleted and whenIsActive priority

godenfreemans, 2022-04-22:
I write those
```
      val lp01 = new StateDelay(2)
      lp01
        .whenIsActive(tx_data_n.set())
        .whenCompleted {
          tx_data_n.clear()
          goto(end)
        }
```
then i get
```
        `esc_entry_fsm_enumDefinition_binary_sequential_esc_entry_fsm_lp01 : begin
          if(when_State_l228_1) begin
            tx_data_n <= 1'b0;
          end
          tx_data_n <= 1'b1;
        end
```
Maybe whenCompleted logic  have more priority than whenIsActive?

Dolu1990, 2022-04-22:
I think things are as they are now, because whenCompleted is used as a exit condition. So something comming after isActive / having more priority

godenfreemans, 2022-04-22:
> 我认为事情就像现在一样，因为 whenCompleted 被用作退出条件。所以在 isActive / 有更多优先级之后出现的东西

👌

closed 2022-04-22 by godenfreemans