SpinalHDL/SpinalHDL #698: Cannot resolve symbol libraryDependencies

0806gcx, 2022-04-26:
I want to use Intellij to configure, and I see this problem in the build. SBT file, what should I do？
`import sbt.Keys.libraryDependencies

ThisBuild / version := "0.1.0-SNAPSHOT"

ThisBuild / scalaVersion := "2.11.12"

val spinalVersion = "1.6.4"
val spinalCore = "com.github.spinalhdl" %% "spinalhdl-core" % spinalVersion
val spinalLib = "com.github.spinalhdl" %% "spinalhdl-lib" % spinalVersion
val spinalIdslPlugin = compilerPlugin("com.github.spinalhdl" %% "spinalhdl-idsl-plugin" % spinalVersion)

lazy val root = (project in file("."))
  .settings(
    name := "test"
    libraryDependencies ++= Seq(spinalCore, spinalLib, spinalIdslPlugin)

  )
`
Cannot resolve symbol libraryDependencies

Dolu1990, 2022-04-26:
name := "test" 
=> 
name := "test", 
?
I guess that's the issue ^^

0806gcx, 2022-04-26:
> name := "test" => name := "test", ? I guess that's the issue ^^

Thank you !Problem solved!

closed 2022-10-14 by numero-744