SpinalHDL/SpinalHDL #699: FSM `makeInstantEntry()` boot state naming

wswslzp, 2022-04-27:
As document said, using `makeInstantEntry()` can remove the `BOOT` state and start the fsm in a user-defined entry state. However, the generated Verilog code shows that it only replaces my entry state's name with "BOOT". 

In scala, 
```scala
  val cacheFsm = new StateMachine {
//    val idle = new State with EntryPoint
    val idle = makeInstantEntry()
    val flagCompare = new State
    val allocation = new State
    val writeBack = new State
// fsm behaviour...
```

In generated Verilog,
```verilog
  localparam cacheFsm_enumDef_BOOT = 2'd0;
  localparam cacheFsm_enumDef_flagCompare = 2'd1;
  localparam cacheFsm_enumDef_allocation = 2'd2;
  localparam cacheFsm_enumDef_writeBack = 2'd3;
```

The state `idle` is missing and is replaced by `BOOT`. Could it preserve its state name?
Currently using SpinalHDL 1.6.4.

wswslzp, 2022-04-27:
Also, I think the word "enumDef" in these state names is kind of redundant. Is this required for SpinalEnum naming?

Dolu1990, 2022-04-27:
The BOOT thing is now fixed in dev.

> Also, I think the word "enumDef" in these state names is kind of redundant. Is this required for SpinalEnum naming?

This is because the enum definition is stored in the StateMachine in the enumDef field. It may avoid some name clashing.

closed 2022-04-27 by Dolu1990

Mido-sun, 2024-11-12:
> The BOOT thing is now fixed in dev.
> 
> > Also, I think the word "enumDef" in these state names is kind of redundant. Is this required for SpinalEnum naming?
> 
> This is because the enum definition is stored in the StateMachine in the enumDef field. It may avoid some name clashing.

under the never name clashing senerio ,do the "enumf" can be cancelled?

Dolu1990, 2024-11-12:
@Mido-sun done in dev ^^

commit aef06b03 (2022-04-27): StateMachine.makeInstantEntry name fix #699