SpinalHDL/SpinalHDL #712: Running test benches without re-elaboration

numero-744, 2022-05-02:
Hi again!

When discussing about verification languages with designers, they told me that they appreciate when the language is interpreted, or could run without re-elaborating the design, because it allows to modify the stimuli and add assertions and re-run the simulation quickly. This is not a key point compared to other things but I think it is worth to at least discuss this.

If I understand correctly, the test bench currently: generates the Verilog, generates the C++ Verilator program which interacts with Scala; then runs the simulation with stimuli generated directly from Scala.

Therefore, it could be possible for the designer to modify the test bench and just re-run it from Scala without elaborating a Verilog file, without generating and compiling a C++ program. So iterating { test bench modifications + simulation } would be _really_ fast.

It seems that SpinalHDL does the whole process each time the simulation is run. Do you plan to implement an optimized version? And do you think it would be worth implementing this?

jijingg, 2022-05-03:
>It seems that SpinalHDL does the whole process each time the simulation is run. Do you plan to implement an optimized version? And do you think it would be worth implementing this?

According to my experience, spinalhdl is often used to do some Unit-tests. Even System-tests are often not too large, and the compilation time is not too long, which is acceptable, so spinal have recompiled every time. If the system is large enough and the compilation time becomes the critical path, It's make sense to do some improve 

two way: 
1. check DUT modifyed(use some hash way)? if not , use current compiled file,  re-run
2. Use incremental compilation mode of Veirlator/vcs to reduce the compilation time

@numero-744 do you tell me about the scale  of you project , How long does it take to compile it at most？

Dolu1990, 2022-05-04:
> without generating and compiling a C++ program

If it is about SpinalSim using the verilator backend, normaly, as long as the hardware generation / user netlist is fully deteriministic SpinalHDL will reuse a previously compiled verilator cache. to speed up things

For instance : 

```scala
  SimConfig.doSim(new Component {
    out(Delay(in(Bool), 10000))
  }){dut =>
    sleep(10)
  }
```

Will run the first time slowly (~20 seconds of verilator compilation)

```
[Progress] at 0.000 : Elaborate components
[Progress] at 0.647 : Checks and transforms
[Progress] at 1.269 : Generate Verilog
[Done] at 1.747
[Progress] Simulation workspace in /home/miaou/pro/riscv/ooo/NaxRiscv/./simWorkspace/unamed
[Progress] Verilator compilation started
[Progress] Verilator compilation done in 20908.398 ms
[Progress] Start unamed test simulation with seed 550893428
[Done] Simulation done in 5.797 ms
```

While the second time it is running it is much faster (3 seconde verilator setup) : 

```
[Progress] at 0.000 : Elaborate components
[Progress] at 0.302 : Checks and transforms
[Progress] at 0.699 : Generate Verilog
[Done] at 1.093
[Progress] Simulation workspace in /home/miaou/pro/riscv/ooo/NaxRiscv/./simWorkspace/unamed
[Progress] Verilator compilation started
[info] Found cached verilator binaries
[Progress] Verilator compilation done in 3019.385 ms
[Progress] Start unamed test simulation with seed 1665783089
[Done] Simulation done in 4.187 ms
```

So it is in part reducing the issue, but sure, for very large scale design, this can be problematic. I'm not sure how verilator people handle changes for very large design.

jijingg, 2022-05-04:
> If it is about SpinalSim using the verilator backend, normaly, as long as the hardware generation / user netlist is fully deteriministic SpinalHDL will reuse a previously compiled verilator cache. to speed up things

great,  I don't know that spinal  already supports use verilator compile cache when the hdl-file hasn't changed.

Dolu1990, 2022-05-06:
@numero-744 Thanks for the info :)