SpinalHDL/SpinalHDL #716: Cannot run SpinalSim with ScalaTest + Mill

zhutmost, 2022-05-04:
I am porting the build tool of some SpinalSim tests from sbt to mill. However, I find that I cannot successfully run `mill mylib.test` in command line. 

I have constructed a small case to reproduce this issue: https://github.com/zhutmost/spinalhdl-template . Just clone it and `mill _.test` . 
Software and environment:
- SpinalHDL and ScalaTest versions, 1.7.0 and 3.2.11, are specified in `build.sc`. (latest).
- Mill version is 0.10.3 (latest).
- Command line: macOS + zsh. 
- Simulation backend: Verilator verison is 4.220.
- JDK version is OpenJDK 11.

Here are some phenomena I observed.

1) Run `mill mylib.test` / `mill _.test` / `mill mylib.test.test` / `mill mylib.test.testOnly` in command line. They have the same output: 
[image]

2) Run tests in IntelliJ IDEA (Click the green arrow in IDE GUI), and everything is OK. IDEA can build the mill project via BSP (Build Server Protocol).
The result is: 
```log
/usr/local/Cellar/openjdk@11/11.0.15/libexec/openjdk.jdk/Contents/Home/bin/java "-javaagent:/Users/zhutmost/Library/Application Support/JetBrains/Toolbox/apps/IDEA-U/ch-1/221.5080.210/IntelliJ IDEA.app/Contents/lib/idea_rt.jar=59478:/Users/zhutmost/Library/Application Support/JetBrains/Toolbox/apps/IDEA-U/ch-1/221.5080.210/IntelliJ IDEA.app/Contents/bin" -Dfile.encoding=UTF-8 -classpath "/Users/zhutmost/Library/Application Support/JetBrains/Toolbox/apps/IDEA-U/ch-1/221.5080.210/IntelliJ IDEA.app.plugins/Scala/lib/runners.jar:/Users/zhutmost/Workspace/spinal/spinalhdl-template/out/mylib/test/compile.dest/classes:/Users/zhutmost/Workspace/spinal/spinalhdl-template/out/mylib/compile.dest/classes:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/net/openhft/affinity/3.21ea1.1/affinity-3.21ea1.1.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/jetbrains/annotations/19.0.0/annotations-19.0.0.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/commons-io/commons-io/2.4/commons-io-2.4.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/jline/jline/3.21.0/jline-3.21.0.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/net/java/dev/jna/jna/5.9.0/jna-5.9.0.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/net/java/dev/jna/jna-platform/5.5.0/jna-platform-5.5.0.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/com/github/oshi/oshi-core/5.2.0/oshi-core-5.2.0.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scala-lang/scala-compiler/2.13.8/scala-compiler-2.13.8.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scala-lang/scala-library/2.13.8/scala-library-2.13.8.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scala-lang/scala-reflect/2.13.8/scala-reflect-2.13.8.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/com/github/scopt/scopt_2.13/3.7.1/scopt_2.13-3.7.1.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/slf4j/slf4j-api/1.7.30/slf4j-api-1.7.30.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/slf4j/slf4j-simple/1.7.25/slf4j-simple-1.7.25.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/com/lihaoyi/sourcecode_2.13/0.2.7/sourcecode_2.13-0.2.7.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/com/github/spinalhdl/spinalhdl-core_2.13/1.7.0/spinalhdl-core_2.13-1.7.0.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/com/github/spinalhdl/spinalhdl-idsl-payload_2.13/1.7.0/spinalhdl-idsl-payload_2.13-1.7.0.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/com/github/spinalhdl/spinalhdl-idsl-plugin_2.13/1.7.0/spinalhdl-idsl-plugin_2.13-1.7.0.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/com/github/spinalhdl/spinalhdl-lib_2.13/1.7.0/spinalhdl-lib_2.13-1.7.0.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/com/github/spinalhdl/spinalhdl-sim_2.13/1.7.0/spinalhdl-sim_2.13-1.7.0.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scala-lang/modules/scala-xml_2.13/2.0.1/scala-xml_2.13-2.0.1.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalactic/scalactic_2.13/3.2.11/scalactic_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-compatible/3.2.11/scalatest-compatible-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-core_2.13/3.2.11/scalatest-core_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-diagrams_2.13/3.2.11/scalatest-diagrams_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-featurespec_2.13/3.2.11/scalatest-featurespec_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-flatspec_2.13/3.2.11/scalatest-flatspec_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-freespec_2.13/3.2.11/scalatest-freespec_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-funspec_2.13/3.2.11/scalatest-funspec_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-funsuite_2.13/3.2.11/scalatest-funsuite_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-matchers-core_2.13/3.2.11/scalatest-matchers-core_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-mustmatchers_2.13/3.2.11/scalatest-mustmatchers_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-propspec_2.13/3.2.11/scalatest-propspec_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-refspec_2.13/3.2.11/scalatest-refspec_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-shouldmatchers_2.13/3.2.11/scalatest-shouldmatchers_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest-wordspec_2.13/3.2.11/scalatest-wordspec_2.13-3.2.11.jar:/Users/zhutmost/Library/Caches/Coursier/v1/https/repo1.maven.org/maven2/org/scalatest/scalatest_2.13/3.2.11/scalatest_2.13-3.2.11.jar" org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner -s mylib.MyTopSimSpec -showProgressMessages true
Testing started at 02:13 ...
[Runtime] SpinalHDL v1.7.0    git head : eca519e78d4e6022e34911ec300a432ed9db8220
[Runtime] JVM max memory : 8192.0MiB
[Runtime] Current date : 2022.05.05 02:13:21
[Progress] at 0.000 : Elaborate components
[Progress] at 0.945 : Checks and transforms
[Progress] at 1.034 : Generate Verilog
[Done] at 1.116
[Progress] Simulation workspace in /Users/zhutmost/Workspace/spinal/spinalhdl-template/./simWorkspace/MyTop
[Progress] Verilator compilation started
[info] Found cached verilator binaries
[Progress] Verilator compilation done in 3655.451 ms

[Progress] Start MyTop a MyTop component should count continuously 0 to 255 simulation with seed 347750660
[Done] Simulation done in 40.278 ms

Process finished with exit code 0
```
The first line in the above log is the command executed by IDEA. 

\\
I find a workaround (implement my only test function) (Thanks to the help from @pwang7 ).
```
    def testSim(args: String*) = T.command {
      super.runMain("org.scalatest.run", args: _*)
    }
```
but this workaround cannot fix the issue of builtin test APIs (test/testOnly/...)

A similar issue is reported as #316 two years ago, but it is not followed later.

zhutmost, 2022-05-04:
When I remove SpinalSim from the testbench, the bug disappeared.
The corresponding test bench is:
```scala
package mylib

import org.scalatest._
import org.scalatest.flatspec._
import spinal.core.sim._ // unused

import scala.util.Random

/** MyTop's testbench, based on ScalaTest */
class MyTopSimSpec extends FixtureAnyFlatSpec with fixture.TestDataFixture {
  val clockPeriod = 10 // No less than 2

  "a MyTop component" should "count continuously 0 to 255" in { td =>
    assert(1 == 1)
  }
}
```
Now the output is:
[image]

So I believe that the above issue is relevant to SpinalSim (at least partially).

Dolu1990, 2022-05-05:
That's some weird issue :/
I can reproduce it in linux Mint.

pwang7, 2022-05-05:
I have the same issue. I cannot run `mill _.test` or `mill _.testOnly <TEST CASE NAME>`:
```
./mill myproj.test.testOnly mylib.MyTopLevelSimTest
[70/70] myproj.test.testOnly 
MyTopLevelSimTest:
[Runtime] SpinalHDL v1.7.0    git head : eca519e78d4e6022e34911ec300a432ed9db8220
[Runtime] JVM max memory : 12008.0MiB
[Runtime] Current date : 2022.05.05 18:28:43
[Progress] at 0.000 : Elaborate components
[Progress] at 0.132 : Checks and transforms
[Progress] at 0.188 : Generate Verilog
[Done] at 0.219
[Progress] Simulation workspace in /home/pwang/Downloads/code/SpinalTemplateMill/./simWorkspace/MyTopLevel
[Progress] Verilator compilation started
Line Number->2
code->compiler.err.doesnt.exist
Message->package spinal.sim does not exist
Source->spinal.sim.DynamicCompiler$InMemoryJavaFileObject[string:///wrapper_verilator/VerilatorNative.java]
 
Line Number->4
code->compiler.err.cant.resolve
Message->cannot find symbol
  symbol: class IVerilatorNative
Source->spinal.sim.DynamicCompiler$InMemoryJavaFileObject[string:///wrapper_verilator/VerilatorNative.java]
 
- Simulate *** FAILED ***
  java.lang.NullPointerException: Cannot invoke "java.lang.Class.newInstance()" because the return value of "spinal.sim.VerilatorBackend.nativeImpl()" is null
  at spinal.sim.VerilatorBackend.<init>(VerilatorBackend.scala:721)
  at spinal.core.sim.SpinalVerilatorBackend$.apply(SimBootstraps.scala:139)
  at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:884)
  at spinal.core.sim.SpinalSimConfig.compileCloned(SimBootstraps.scala:818)
  at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:795)
  at mylib.MyTopLevelSimTest.$anonfun$new$1(MyTopLevelSimTest.scala:11)
  at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.scala:18)
  at org.scalatest.OutcomeOf.outcomeOf(OutcomeOf.scala:85)
  at org.scalatest.OutcomeOf.outcomeOf$(OutcomeOf.scala:83)
  at org.scalatest.OutcomeOf$.outcomeOf(OutcomeOf.scala:104)
  ...
- An empty Set should have size 0
- Invoking head on an empty Set should produce NoSuchElementException
1 targets failed
myproj.test.testOnly 1 tests failed: 
  mylib.MyTopLevelSimTest Simulate
```

I have to define my own `testSim` to run each testcase:
https://github.com/pwang7/SpinalTemplateMill/blob/5891e894b8e59ee8a98fa69478843572b4753a02/run.sh#L23

zhutmost, 2022-06-07:
This issue can also be reproduced with [uTest](https://github.com/com-lihaoyi/utest), another test framework.
So I guess this issue has nothing to do with ScalaTest. 

Just paste my uTest spec code here, if anyone needs.
```scala
package mylib

import spinal.core.sim._
import utest._

import scala.util.Random

/** MyTop's testbench, based on ScalaTest */
object MyTopSimSpec extends TestSuite {
  val tests = Tests {
    val clockPeriod = 10 
    val compiled = SimConfig
      .withConfig(MySpinalConfig)
      .allOptimisation
      .compile(new MyTop)                   // <- It fails here.

    test("increments continuously") {
      compiled.doSim { dut =>
        // ...
      }
    }
  }
}

```

huaixv, 2022-06-08:
It seems a classpath problem. Mill have a mechanism to separate the CP used by testers and and by Mill itself, see [here](https://github.com/com-lihaoyi/mill/blob/main/main/util/src/mill/util/Jvm.scala#L16).

`DynamicCompiler` and `URLClassLoader`  gets the Mill CP from `System.getProperty("java.class.path")`, so they cannot find `spinal.sim` package.

[Here](https://github.com/huaixv/SpinalHDL/commit/f28c208a1c5adb19c3a09433a31cbba0b6cad3e1) is a dirty hack to make it work on my machine, but not fully tested yet.

pwang7, 2022-06-08:
> It seems a classpath problem. Mill have a mechanism to separate the CP used by testers and and by Mill itself, see [...]

Nice catch.
Could you please test your patch and submit a PR?

closed 2022-06-10 by Dolu1990