SpinalHDL/SpinalHDL #717: Axi4SlaveFactory violation of BVALID vs BREADY rule

sebastien-riou, 2022-05-05:
I suspect Axi4SlaveFactory violates [the write transaction rule about BVALID vs BREADY](https://developer.arm.com/documentation/ihi0022/e/AMBA-AXI3-and-AXI4-Protocol-Specification/Single-Interface-Requirements/Relationships-between-the-channels/Dependencies-between-channel-handshake-signals?lang=en#CIHJEBCI) "the slave must not wait for the master to assert BREADY before asserting BVALID":
I see the following path: io_cpuBus_b_valid <- mapper_writeRsp_rValid <- mapper_writeRsp_valid <- mapper_writeJoinEvent_fire_1 <- mapper_writeJoinEvent_ready <- mapper_writeRsp_ready <- mapper_writeRsp_m2sPipe_ready <- io_cpuBus_b_ready

I see this using ``sbt "runMain mylib.testAxiSlaveGen"`` with the attached project
[SpinalTemplateSbt.zip](https://github.com/SpinalHDL/SpinalHDL/files/8629366/SpinalTemplateSbt.zip)

extract of the relevant verilog:
````
assign io_cpuBus_b_valid = mapper_writeRsp_m2sPipe_valid;
assign mapper_writeRsp_m2sPipe_valid = mapper_writeRsp_rValid;
always @(posedge clk or posedge reset) begin
    if(reset) begin
      ...
      mapper_writeRsp_rValid <= 1'b0;
    end else begin
      ...
      if(mapper_writeRsp_ready) begin
        mapper_writeRsp_rValid <= mapper_writeRsp_valid;
      end
    end
end
always @(*) begin
    if(io_cpuBus_w_payload_last) begin
      mapper_writeJoinEvent_ready = (mapper_writeRsp_ready && (! mapper_writeHaltRequest));
    end else begin
      mapper_writeJoinEvent_ready = (! mapper_writeHaltRequest);
    end
end
assign mapper_writeJoinEvent_fire_1 = (mapper_writeJoinEvent_valid && mapper_writeJoinEvent_ready);
always @(*) begin
    if(io_cpuBus_w_payload_last) begin
      mapper_writeRsp_valid = mapper_writeJoinEvent_fire_1;
    end else begin
      mapper_writeRsp_valid = 1'b0;
    end
end

always @(*) begin
  mapper_writeRsp_ready = mapper_writeRsp_m2sPipe_ready;
  if(when_Stream_l368) begin
    mapper_writeRsp_ready = 1'b1;
  end
end
assign mapper_writeRsp_m2sPipe_ready = io_cpuBus_b_ready;
assign when_Stream_l368 = (! mapper_writeRsp_m2sPipe_valid);

````

Dolu1990, 2022-05-05:
I think it is fine, The reasons is : 

```
      if(mapper_writeRsp_ready) begin
        mapper_writeRsp_rValid <= mapper_writeRsp_valid;
      end
```

mapper_writeRsp_ready is the front of the stage, which is always True if the stage isn't loaded with a transaction (when_Stream_l368).

So that way we never wait for the output bready to be true if there is nothing on the output

From an abstract perspective the axi.B is driven by a m2spip

Dolu1990, 2022-05-06:
So, in other words, from a purly combinatorial perspective, yes, there is a relation ship, but from a transactional perspective, a transaction on the B channel of the axiSlaveFactory can always come  if the buffer is empty => (doesn't wait on B ready)

Does it make sense ?

asicdruide, 2022-05-06:
As for all flow controlled interfaces:
 * the sender flags the availability of new data (BVALID in this case), this is purely based on local data
 * the receiver flags ability to accept at least one data (BREADY in this case), this is purely based on local data
 * consensus between sender and receiver is that both transfer responsibility for one data from sender to receiver based on the same condition (BVALID=='1' and BREAD=='1'). 

Both sides evaluate this same condition (in both cases based on one local and one remote signal). The sender discards the data when condition is true, receiver captures the data if condition is true. This means on the same clock edge data is actually transferred. Please note that both local signals (BVALID for sender and BREADY for receiver) do not depend on each other. 

This is how all flow controlled interfaces (like AXI) work.

kleinai, 2022-05-07:
I'm invested only in making sure the bus factory is AXI compliant.

Would something like this work? (I'm on windows so pulling up the tests is a challenge)
```
  var writeCmd = bus.writeCmd.unburstify
  val writeJoinEvent = StreamJoin.arg(writeCmd, bus.writeData).haltWhen(writeHaltRequest)
  val writeRsp = Stream(Axi4B(bus.config))
  bus.writeRsp << writeRsp.stage()
  writeJoinEvent.ready := !bus.writeRsp.valid || bus.writeRsp.ready
  when(bus.writeData.last) {
    writeRsp.valid := writeJoinEvent.fire
  } otherwise {
    writeRsp.valid := False
  }
```

Break the link between writeRsp valid and halt writing if there's a response in the pipeline stage.

asicdruide, 2022-05-07:
Sorry i can't read spinalHDL. So i can't extract the behavior from the code snippet.
What i've described in my last post is the quintecence of all mistakes i've either made myself or reverse engineered in others code. 
The sender knows when it has data available and puts it on the bus (BVALID:='1' and BRESP:=<whatever>). 
Afterwards the sender observes the bus for condition (BVALID=='1' and BREADY=='1'). 
If condition is false, keep BVALID and BRESP stable. 
If codition is true the sender has the choise to either keep BVALID and update BRESP for the next transfer (back-to-back) or set BVALID:='0'. 

The receiver does similar things. 
If it can accept a data (independent whether there is one or not) it sets BREADY:='1'. 
Afterwards receiver observes the bus for condition (BVALID=='1' and BREADY=='1', the contract with sender).
Once condition is true, BRESP is captured and receiver can decide. 
If more data can be accepted, keep BREADY. 
If not, set BREADY:='0' until we can again accept a data.  

Both ends take their decision to set BVALID resp BREADY to '1' completely based on local data. 
The remote signal BREADY resp BVALID are just taken into account to finalize the pending transfer. 

The key is the contract condition that both ends of the interface act on the same clock edge doing complementary things.

sebastien-riou, 2022-05-07:
Thanks all for your comments, for now I am convinced by @Dolu1990 explanations and I think I shouldn't have created this issue. I am not totally sure because my real design does show the issue but it is using out of date SpinalHDL. I am in the process of porting it to SpinalHDL 1.7, I will close the issue when this is done and I see that the issue is gone.
Side note: I created the issue after checking it using a minimal design with SpinalHDL 1.7 but unfortunately I made a mistake in the testbench which made it look like the issue is there (forgot to set LAST signal).

Dolu1990, 2022-05-07:
@sebastien-riou You are right, older version of the axi slave factory did had a issue : 

https://github.com/SpinalHDL/SpinalHDL/commit/38ba5eb3fdc8ab0b7f26514b1869645a098011c1

No worries ^^

sebastien-riou, 2022-05-09:
Real project using SpinalHDL 1.7 works, closing this issue, sorry.

closed 2022-05-09 by sebastien-riou

Dolu1990, 2022-05-09:
No worries ^^