SpinalHDL/SpinalHDL #719: all signal name _zz_ when use 1.7.0 in Linux

aiit-linry, 2022-05-05:
Its OK in windows or use 1.6.4 in Linux, include centos 7.9 and ubuntu 18.04

aiit-linry, 2022-05-05:
When using scala version > 2.13

aiit-linry, 2022-05-05:
1.7.0+2.12.x = compile failed,
[image]

1.7.0+2.13.x = zz in names

1.7.0+2.11.x = everything OK , but i need Random.nextLong function in 2.13.x :(

hope fix this issue

tanhongze, 2022-05-09:
I use Linux with Spinal 1.7.0 and scala 2.13.8 too.
However, everything seems OK in my environment. 
What's the different?

Dolu1990, 2022-05-09:
I noticed that when i was updating project which were already compiled once with a older version, i had some issue.

Did you tried "sbt clean" ?

aiit-linry, 2022-05-11:
> I use Linux with Spinal 1.7.0 and scala 2.13.8 too. However, everything seems OK in my environment. What's the different?

[image]
[image]

Are there any errors in my configuration? I tried template and only modify scala version to 2.13.8.

aiit-linry, 2022-05-11:
> I noticed that when i was updating project which were already compiled once with a older version, i had some issue.
> 
> Did you tried "sbt clean" ?

yeap, I tried "sbt clean", it seems doesnt work. :(

aiit-linry, 2022-05-11:
But Its OK in my windows env, that is really confusing.

Dolu1990, 2022-05-11:
I just tested on my side, Scala 2.13.8, all seems fine.

Which version of java / javac do you have ? (--version)

aiit-linry, 2022-05-12:
> I just tested on my side, Scala 2.13.8, all seems fine.
> 
> Which version of java / javac do you have ? (--version)

javac 1.8.0_312 in my ubuntu and javac 1.8.0_262 in my centos

numero-744, 2022-10-20:
Hi @aiit-linry,
Is the issue still here?

Else have you tried `sbt clean` as Dolu mentioned?

aiit-linry, 2022-10-21:
> Hi @aiit-linry, Is the issue still here?
> 
> Else have you tried `sbt clean` as Dolu mentioned?

yes, it is still in 1.7.0, I have tried 'sbt clean' but it doesnt work. But when I update spinal version to 1.7.2, all is well in my ubuntu and my centos. @numero-744 
should I close this issue under this circumstance? @Dolu1990

numero-744, 2022-10-21:
I close it. Thanks for the report

closed 2022-10-21 by numero-744