SpinalHDL/SpinalHDL #723: forkSensitive's behavior is different between verilator and vcs

finalex, 2022-05-09:
When i use spinalsim to simulate my code, i write some code as below:

`forkSensitive2(io.bus.rd_req, io.bus.rd_ptr) {
    io.bus.rd_stm #= Random.nextInt() & 0xff
}
`
io.bus.rd_req, io.bus.rd_ptr are outputs of dut. 
io.bus.rd_stm is input of dut.

What i except is the io.bus.rd_stm changes randomly whenever io.bus.rd_req or io.bus.rd_ptr changed the same time.

If i use verilator as backend, the waveform shows what i expected.
[image]

But in vcs, it's strange. io.bus.rd_stm changes half cycle later.
[image]

I'm not sure if it is the delta cycle problem.

finalex, 2022-05-09:
More info:

I expand the delta cycle in verdi, I see the delta cycle mode is  active -> nba -> active.

I tried to print value of rd_req, rd_ptr and other signals each delta cycle by using forkSensitive {}. 

Print works well for the signal changes only in the first two region, if signal changes in the last active region, value will be print in the next simulation time.

Dolu1990, 2022-05-09:
@Name1e5s

I'm not personnaly familiar with VCS / VPI stuff, any idea ? I agree it could be delta cycle issue, or it maybe some write being buffered.
If you want we can do a call to talk about the technicality of SpinalSim

Dolu1990, 2022-05-09:
ok ^^
Let's me know how it goes.