SpinalHDL/SpinalHDL #724: %t instead of %d

hanm2019, 2022-05-09:
https://github.com/SpinalHDL/SpinalHDL/blob/734e31103e1731b557a7ec05c743e7bd8a73014a/core/src/main/scala/spinal/core/internals/ComponentEmitterVerilog.scala#L651

when I try to display some info by report(Seq(REPORT_TIME, "message")), the verilog translate by spinalHDL is 
```
$display("NOTE %dmessage", $time); // core.scala:L475 
```
in my opinion, use %t is better than %d for $time type

Dolu1990, 2022-05-09:
I think you are right. I pushed the fix.

Still there is a weird thing. On my end : 

```scala
  SimConfig.withWave.compile(new Component {
    val fsm = new StateMachine{
      val IDLE = makeInstantEntry()
      val STATE_A, STATE_B, STATE_C = new State
      
      IDLE.whenIsActive(goto(STATE_A))
      STATE_A.whenIsActive(goto(STATE_B))
      STATE_B.whenIsActive(goto(STATE_C))
      STATE_C.whenIsActive {
        report(L"Miaou $REPORT_TIME")
        goto(STATE_B)
      }
    }
  }).doSim{ dut =>
    dut.clockDomain.forkStimulus(10)
    dut.clockDomain.waitSampling(100)
  }
```

is reporting weird time as : 

```
[Progress] Start unamed test simulation with seed 253137368
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                    0
NOTE Miaou                 1000
NOTE Miaou                 1000
NOTE Miaou                 1000
NOTE Miaou                 1000
NOTE Miaou                 1000
NOTE Miaou                 1000
NOTE Miaou                 1000
NOTE Miaou                 1000
```

Do you have the same kind of issue ?

hanm2019, 2022-05-09:
I have the same kind of issue
```
NOTE               546000add 1
NOTE               546000add 1
NOTE               546000add 1
NOTE               546000add 1
NOTE               546000add 1
NOTE               546000add 1
NOTE               546000add 1
```

hanm2019, 2022-05-09:
maybe cause by the `timescale 1ns/1ps`?

Dolu1990, 2022-05-09:
Ahhh you are likely right ^^

Dolu1990, 2022-05-10:
I just tested, yes this is related to the time scale stuff

Dolu1990, 2022-05-10:
Else all good ?

hanm2019, 2022-05-11:
all good

closed 2022-05-11 by hanm2019