SpinalHDL/SpinalHDL #725: Operator `-|` (subtraction by sat carry bit) is incorrect?

tanhongze, 2022-05-09:
Substraction with saturation should get the min value.
However, in Spinal, `U(0, 2 bits) -| U(1)` would got U(3, 2 bits) instead of U(0, 2 bits).
It is saturated to max value.
Is this a bug?

Dolu1990, 2022-05-10:
Yes you are right i think.
@jijingg 
Seems releated to https://github.com/SpinalHDL/SpinalHDL/blob/dev/core/src/main/scala/spinal/core/UInt.scala#L120
which is used by the -| operator.
Seems like we should have a specific implementation for substraction on UInt instead, right ?

jijingg, 2022-05-10:
> Seems like we should have a specific implementation for substraction on UInt instead, right ?

yes，it' quit special ,   A specific implemetation is added to deal with this case.

commit 89c4d164 (2022-05-10): Merge pull request #728 from jijingg/sat

fix #725
commit 985497d7 (2022-05-10): fix #725
closed 2022-05-10 by Dolu1990