SpinalHDL/SpinalHDL #729: Is setAsReg stable / safe for user use?

numero-744, 2022-05-11:
Hi!

I am implementing a protocol converter and it seems that having registers on the output of bundles is useful. I am doing things like `io.protocol.PORT.setAsReg()`. Is it safe?

I am wondering because `setAsReg()` [is not in the documentation](https://spinalhdl.github.io/SpinalDoc-RTD/master/search.html?q=setasreg&check_keywords=yes&area=default).

If it is safe I could create a PR to document it.

numero-744, 2022-05-11:
Plus a question:

If I use `.setAsReg()` inside an `ClockedArea`, will the register be driven by the clock of the component, or the one of `ClockedArea`?

Dolu1990, 2022-05-11:
>  Is it safe?

Yes it is. It is mostly lowlevel API hidden behind Reg(xxx) : 
https://github.com/SpinalHDL/SpinalHDL/blob/34796a8279dd6d7c7d6513e57effdb050079e103/core/src/main/scala/spinal/core/Reg.scala#L35

There is no weird side effect using it.

> If it is safe I could create a PR to document it.

yes, Thanks :)
The feature is mostly usefull for after the fact patch / fixes / abstract stuff.

> If I use .setAsReg() inside an ClockedArea, will the register be driven by the clock of the component, or the one of ClockedArea?

It will use the clock in which the original signal was defined, so if the signal was defined outside the clocking area it will use the component clock

numero-744, 2022-05-11:
> It will use the clock in which the original signal was defined

This is what I expected.

Thanks for your answers, I will send a PR to the documentation repository in the following days.

numero-744, 2022-05-28:
Done: https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Sequential%20logic/registers.html#transforming-a-wire-into-a-register

(Triaging my own issues)

closed 2022-05-28 by numero-744