SpinalHDL/SpinalHDL #730: Custom encoding for FSM states

numero-744, 2022-05-11:
Hello,

I managed to apply a SpinalHDL-provided encoding to states of a state machine, but I could not apply a custom one (with a list of `STATE -> value`).

Maybe it is possible to take a custom `SinalEnum`, apply a custom encoding on it and then give it as states for a state machine?

Dolu1990, 2022-05-11:
It isn't very much tested, but there is the setEncoding function that you can use, is it what you used ?
How did you tried it ?

numero-744, 2022-05-11:
This does not work because of types:

```scala
val IDLE = State()

setEncoding(SpinalEnumEncoding(
      IDLE -> 0
    ))
```

`(spinal.lib.fsm.State, Int)`, expected `(BigInt, BigInt)`

I also tried by wrapping `IDLE` into `enumOf()` and `stateToEnumElement()` and it also does not work.

Dolu1990, 2022-05-12:
I just pushed a fix in dev which allows : 

```scala
  SimConfig.withFstWave.compile(new Component {
    val fsm = new StateMachine{
      val IDLE = new State
      val STATE_A, STATE_B, STATE_C = new State

      setEntry(IDLE)

      IDLE.whenIsActive(goto(STATE_A))
      STATE_A.whenIsActive(goto(STATE_B))
      STATE_B.whenIsActive(goto(STATE_C))
      STATE_C.whenIsActive(goto(STATE_B))

      setEncoding(SpinalEnumEncoding(id => id*2))

    }
  }).doSim{ dut =>
    dut.clockDomain.forkStimulus(10)
    dut.clockDomain.waitSampling(100)
  }
  `` 
  
  => 
  
  ```verilog
    localparam fsm_enumDef_BOOT = 4'd0;
  localparam fsm_enumDef_IDLE = 4'd2;
  localparam fsm_enumDef_STATE_A = 4'd4;
  localparam fsm_enumDef_STATE_B = 4'd6;
  localparam fsm_enumDef_STATE_C = 4'd8;
  ```
  
An alternative : 

```scala
      val mapping = mutable.LinkedHashMap[Int , BigInt](
        states.indexOf(IDLE) -> 0,
        states.indexOf(STATE_A) -> 1,
        states.indexOf(STATE_B) -> 3,
        states.indexOf(STATE_C) -> 7,
        states.indexOf(stateBoot) -> 15
      )
      setEncoding(SpinalEnumEncoding(mapping.apply))
``` 
  
But i added a shorter way just now as : 

```scala
      setEncoding(
        IDLE -> 0,
        STATE_A -> 1,
        STATE_B -> 3,
        STATE_C -> 7,
        stateBoot -> 15
      )
```  

<3

numero-744, 2022-05-12:
I have tried the alternative with mapping but I could not make it work because `indexOf()` returns an `Int` instead of a `BigInt`.

The shorter way you added seems great :D

Will it work with values such as `B"1010"`? I ask because for FSM it is better to do gray-like encodings. If the FSM is just a loop, a gray encoding is great (I think gray encoding is not in [the native encodings](https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Data%20types/enum.html#encoding), maybe I could try to add it) but for more complex state machines I think it is better to use a custom encoding, and it would be easier for the designer to work with binary than with decimal.

Jeff-Ciesielski, 2022-06-03:
Before I duplicate effort, anyone mind if I take a crack at this? Trying to get more familiar with Spinal's innards and this seems pretty reasonable and useful for folks.

numero-744, 2022-06-04:
> Will it work with values such as `B"1010"`?

This is related to https://github.com/SpinalHDL/SpinalHDL/issues/161

Dolu1990, 2022-06-06:
@Jeff-Ciesielski Sure, go ahead :)

Just one tip, first think how the user API should look like ^^

numero-744, 2022-10-14:
@Jeff-Ciesielski did you manage to implement it or design the API?

Jeff-Ciesielski, 2022-10-14:
@numero-744 Unfortunately I did not, been swamped at work :(

numero-744, 2022-10-26:
So let's say that:

* magically generating gray-like encoding for FSM is not planned (won't open an issue, quite hard to define because it requires to know which transitions are taken more often to optimize the encoding)
* Gray encoding as a default encoding is useful, opened #939 for this.
* Using Bits to construct Encoding has already been mentioned: #161

closed 2022-10-26 by numero-744