SpinalHDL/SpinalHDL #735: [Feature Request] StreamTransactionExtender, controlled by logic rely on payload, instead of a external counter

ngzh, 2022-05-13:
Stream API is awesome, yet to my knowledge it lacks of a one-to-many version of `translateFrom` sort of stuff. 

I. What I want
It's kind of like `StreamTransactionExtender`(STE for short), but the amount and content of `output.payload` is determined by logic which depends on the content of `input.payload`, instead of a external counter in STE. Example usage may like the op to micro-op decoding, and case below

II. What I've done
I've wrote a draft of what I meant, based on the current STE implementation, and a simple test bench, which drive it with a Stream of 4bit UInt `n`, and expected output like `n, n-1, ..., 0`, as waveform shown below. 

[image]

link to complete runnable code which can be paste into `SpinalTemplateSbt` repo and run, [gist](https://gist.github.com/ngzh/37d7c9eef6fc7128f5abe9a949374c16), sample code using it shown below. `8` is the width of the counter, `last` indicate the end of transaction, `cnt` is a counter represent the number of output packet in a transaction, `advance` just means `output.fire`

```scala
class MyTopLevel extends Component {
  val dt = UInt(4 bits)
  val io = new Bundle {
    val push = slave Stream(dt)
    val pop = master Stream(dt)
  }

  val ste = new StreamTransactionExtender2(io.push.payloadType, io.pop.payloadType, 8,
    (out_pld : UInt, last: Bool, in_pld: UInt, cnt: UInt, advance: Bool) => {
      val cnt_down_r = Reg(dt)
      when(cnt === 0) {
        cnt_down_r := in_pld - 1
      } elsewhen(advance) {
        cnt_down_r := cnt_down_r - 1
      }

      out_pld := Mux(cnt === 0, in_pld, cnt_down_r)

      last := in_pld === cnt.resized
    })

  ste.io.input << io.push//.queueLowLatency(3)
  ste.io.output >> io.pop
}
```
III. Questiones

0. Is there existing stuff that fulfill my need? I've read the source code as much as I can(Stream Fragment etc.), but the closet stuff I found was STE. 

If no, maybe there's chance for me to do some contribution to SpinalHDL, yi-ha. Could you please review my draft little bit
1. ` (out_pld : UInt, last: Bool, in_pld: UInt, cnt: UInt, advance: Bool) ` is the parameter of the dataAssign lambda proper? I just put signals to drive like (`out_pld`, `last`) to left most. 
2. What do you think of the counter in the dataAssign lambda? It's kind of wired but in my use case it's pretty useful.
3. I would like to use it to make a one-to-many version of `translateFrom`/`translateInto`, what name do you think is consist with existing API? `translateTransactionFrom` / `translateFragmentFrom`/`translate1toNFrom`/`extendFrom`    `translateIntoTransaction`/`extendInto`?

Thanks for your time and the amazing SpinalHDL

kleinai, 2022-05-13:
If I'm understanding correctly, you want to accept one beat of stream data and generate some number of output beats based on that input.

There's `StreamFragmentGenerator` which accepts an event and generates beats based on a `Vec`. Maybe you could expand that into a dynamic form which uses a translator and control function like you have in your example. I think the `Generator` name would be more appropriate for your use case?

Readon, 2022-05-13:
Why do you want to mix those logic?
As I understand, it can be split into two parts.
1. use StreamTransactionExtender to extend input stream.
2. create the destination payload of output stream by translateWith function.

ngzh, 2022-05-13:
@kleinai @kleinai thanks for the reply. 

I may not have made the use case clear, the output payload is determined by **the content** of input payload.  Say there is a load DMA controller, which takes a dma request `(addr, size)` which size can be arbitrary, but mem controller may only accept uop read size within, say, 64 byte, so if there is an dma read request like read 254 from addr 10000: `(10000, 254)`, the desired uop sequence to mem controller should be like `(10000, 64) (10064, 64) (10128, 64) (10192, 62)`. (For real use case, read controller may also request `addr` to be a multiple of 64, so the `dataAssignment` logic can be more complicate than a simple counter can handle)

My understanding on what `StreamFragmentGenerator` does is to *serialize* an existing large data packet with given size; use `StreamTransactionExtender` to duplicate and `translateWith` to transform have thesome limitation, since the `count` may not accessible to `dataAssignment` lambda of `translateWith`.

kleinai, 2022-05-13:
I think I've understood what you're asking for. The extra example is helpful.
I stand by that we should adapt a new version of `StreamFragmentGenerator` for this purpose. I can take a crack at an example of this later today.

ngzh, 2022-05-13:
@kleinai it's true that the example shown in the waveform is not that obvious, should carry out this extra example in the first place.

Well, in my use case, the output payload type is sort of [flit](https://en.wikipedia.org/wiki/Flit_(computer_networking)), so output stream type is `Stream[XXXFlit]`. Since flit usually contains first/last bit,  `Stream[Fragment[XXXFlit]]` seems functionally and linguistically redundant to me.

kleinai, 2022-05-13:
If you want a `Filt` data type that could be a small converter component from any `Fragment` stream. Like a `.toFlit` enrichment function. This is also similar to Avalon streaming which Spinal does not yet have definitions for.

Readon, 2022-05-14:
pass the count value through payload ？

kleinai, 2022-05-14:
This is very close to what you're already made but seems a little cleaner maybe?

```scala
object StreamGenerator {
  def apply[T <: Data, T2 <: Data](input: Stream[T], bitsWidth: Int)(generator: (Flow[T], UInt, Bool, Bool) => Stream[T2]): Stream[T2] = {
    val counter = Counter(bitsWidth bit)

    val inputStage = input.m2sPipe(holdPayload = true)

    val done = Bool(false)
    done.allowOverride
    val genInputFlow = Flow(inputStage.payload)
    genInputFlow.payload := inputStage.payload
    genInputFlow.valid := inputStage.valid
    val ret = generator(genInputFlow, counter, counter.willOverflowIfInc, done)

    when(ret.fire) {
      when(done) {
        counter.clear()
      } otherwise {
        counter.increment()
      }
    }

    when(counter.willOverflow || counter.willClear) {
      inputStage.ready := True
    } otherwise {
      inputStage.ready := False
    }

    ret
  }
}
```

ngzh, 2022-05-16:
@kleinai thanks for the code, the use of counter as a control signal generator (willClear indicates new input incoming, willIncrement indicates an output will be fire) is inspiring. @Readon may checkout the [extra example](https://github.com/SpinalHDL/SpinalHDL/issues/735#issuecomment-1126336798) give above, it's kind of more than "pass the count value through payload"

I've come up with a new version [link to gist](https://gist.github.com/ngzh/ad09a4e9473b535cc2ff31ad8215f215) (sorry for the gist in the first post in this issue was set to be private [link to gist](https://gist.github.com/ngzh/37d7c9eef6fc7128f5abe9a949374c16)), still a draft, but close to what I want. This issue can be close and the final implementation may come up as a PR, if everything goes well. Thanks again for your time.

------
Example use case in [new version](https://gist.github.com/ngzh/ad09a4e9473b535cc2ff31ad8215f215) is an DMA decoder some dumb DMA Request:

 [`req_id`, `addr`, `size`]  where `addr` and `size` can be arbitrary as `size` >= 1

to some FLIT

 [`req_id`, `flit_id`, `addr`, `size`, `skip`, `first`, `last`]   where `addr` should be aligned to 64, `skip` + `size` <= 64, `skip` >= 0, `size` >= 1

For example, an input DMA request:

 [`addr`=7080, `size`=95] (7080=64*110+40)

 shall be decode into three flit (just leave out `req_id`,`flit_id`, `first`, `last` as their name are pretty self explanatory):

  - [`addr`=7040, `skip`=40, `size`=24]
  -  [`addr`=7104, `skip`=0, `size`=64] 
  -  [`addr`=7168, `skip`=0, `size`=7] .  

To implement DMA decoder described above, code using alternative version of  `StreamTransactionExtender` is given below[L71 in gist](https://gist.github.com/ngzh/ad09a4e9473b535cc2ff31ad8215f215):

```scala
  val ste = new StreamTransactionExtender2(
    io.push.payloadType, io.pop.payloadType, 16 bits,
    (out_pld : DMAFlit, last: Bool, in_pld: DMAReq, in_pld_nxt: DMAReq, cnt: Counter) => {
      in_pld.setName("in_pld"); in_pld_nxt.setName("in_pld_nxt")
      val addr_tosend_r = RegInit(in_pld.addr).setName("addr_tosend_r")
      val size_remain_r = RegInit(in_pld.size).setName("size_remain_r")
      val size_tosend_r = RegInit(out_pld.size).setName("size_tosend_r")
      val skip_tosend_r = RegInit(out_pld.skip).setName("skip_tosend_r")

      when(cnt.willClear) { // on recieve new dma_req
        val addr_next = in_pld_nxt.addr
        val addr_remain = addr_next & U(0x3F).resized
        val size_tosend = Min(64 - addr_remain, in_pld_nxt.size)
        addr_tosend_r := addr_next & ~U(0x3F).resize(addr_next.getBitsWidth)
        size_tosend_r := size_tosend.resized
        skip_tosend_r := addr_remain.resized
        size_remain_r := in_pld_nxt.size - size_tosend
      } elsewhen(cnt.willIncrement) { // emitting more than one flit
        val size_tosend = Min(size_remain_r, U(64).resized)
        addr_tosend_r := addr_tosend_r + 64
        size_tosend_r := size_tosend.resized
        skip_tosend_r := 0
        size_remain_r := size_remain_r - size_tosend.resized
      }

      out_pld.req_id  := in_pld.req_id // RegNext(in_pld_nxt.req_id)
      out_pld.addr    := addr_tosend_r
      out_pld.size    := size_tosend_r
      out_pld.skip    := skip_tosend_r
      out_pld.flit_id := cnt
      out_pld.first   := cnt === 0
      out_pld.last    := size_remain_r === 0

      last := out_pld.last
    })
```
compare to the reference software version used in sim [L137 in gist](https://gist.github.com/ngzh/ad09a4e9473b535cc2ff31ad8215f215), it's pretty close in shape:

```scala
      val ref = scala.collection.mutable.ArrayBuffer[(Int, Int, Int, Int, Int, Boolean, Boolean)]()
      val res = scala.collection.mutable.ArrayBuffer[(Int, Int, Int, Int, Int, Boolean, Boolean)]()
      StreamMonitor(dut.io.push, dut.clockDomain) { p =>
        var size_remain = p.size.toInt

[... 51 lines cut ...]
    counter.increment()
  }

  when(io.input.fire) {
    payload := io.input.payload
    counter.clear()
  }

  val lastValid = Bool()
  // val haltOut   = Bool()
  dataAssignment(io.output.payload, lastValid, payload, io.input.payload, counter)
  // dataAssignment(io.output.payload, lastValid, payload, Mux(io.input.fire, io.input.payload, payload), counter)   // alternative implementaion
  io.first := (counter === 0) & io.output.fire
  io.last := lastValid & io.output.fire
  io.working := io.output.fire
  io.input.ready := (!extState | io.last)
  io.output.valid := extState // && !haltOut
}
```

Further thing to work with:

1. parameter of generator lambda `(out_pld : DMAFlit, last: Bool, in_pld: DMAReq, in_pld_nxt: DMAReq, cnt: Counter)` is pretty messed up, not to mention this lambda can be further use to drive 1) a `halt` signal to pause the generator, 2) a `drop` signal for generate zero output for given input. It seems worth a try, to bundle all these control signal, including `last`/`halt`/`drop` and `nxt payload` and `counter`, to an StreamExtendControl sort of stuff.

2. use above stuff to implement `extendFrom` (similar to `translateFrom`)... anyway to work with existing given Stream API and coding style.

closed 2022-05-16 by ngzh