SpinalHDL/SpinalHDL #744: Have some problems importing precompiled libraries with VCS

jjyy-Huang, 2022-05-21:
Hello,

It's wonderful that SpinalHDL simulation supports VCS now. But there is a question that confuses me. When I simulate an external IP `(e.g. Xilinx FIFO)` with the Blackbox, how should I add the pre-compiled library to link it? I couldn't find the API mentioned in the documentation. The mapping of `synopsys_sim.setup` is also not done by just passing the compile parameter via VCSFlags.

jjyy-Huang, 2022-05-21:
Because the `${workspacePath}` will be cleaned before each time simulation run.  So I run a script that copies `synopsys_sim.setup` to `${workspcePath}` for VCS mapping the pre-compiled simlibrary before vlogan running. And Now I can simulate Xilinx IP by this method.

Maybe it is possible to add a setupFlags in VCSFlags and execute it before the analysis step?

wswslzp, 2022-05-22:
Sorry, I'm an ASIC engineer and not familiar with VCS simulation flow for VIVADO IP. How does VCS use `synopsys_sim.setup` file in a normal three-step flow?

And now SpinalHDL 1.7.0 also supports native VIVADO simulation flow by `withXSim`. Maybe you can try this one.

wswslzp, 2022-05-22:
Maybe adding a parameter `simSetupPath` to `withVCS` that copies `synopsys_sim.setup` file to `workspacePath` would help?

jjyy-Huang, 2022-05-23:
Sorry for the late reply.

Yes, that is what I meant. I just add a small step before `analyzeSource()`.

```   scala
 def myScript(): Unit = {
      s"cp ./synopsys_sim.setup $workspacePath".!
      val displayScript =
        s"""#/bin/bash
           verdi -f filelist.f -ssf $toplevelName.fsdb
           """.stripMargin
      val displayScriptFile = new PrintWriter(new File(
        s"$workspacePath/display.sh"))
      displayScriptFile.write(displayScript)
      displayScriptFile.close()
      s"chmod 777 $workspacePath/display.sh".!
    }

    println(f"[Progress] VCS compilation started.")
    val startAt = System.nanoTime()
    myScript()
    analyzeSource()
    elaborate()
    val deltaTime = (System.nanoTime() - startAt) * 1e-9
    println(f"[Progress] VCS compilation done in $deltaTime%1.3f s")
```
But I think this can do better in your ways.

wswslzp, 2022-05-23:
I opened a PR #747 and you can review it.

closed 2022-05-24 by jjyy-Huang