SpinalHDL/SpinalHDL #745: Implement BitwiseOp for Vec[T]

numero-744, 2022-05-21:
Hi,

Is it possible for `Vec[T]` to implement `BitwiseOp` if `T` implements it?

- [ ] Can Scala represent it?
- [ ] Would it be in accordance with the SpinalHDL paradigm?

If so, I think it would be a good first issue ;)

numero-744, 2022-05-23:
The idea is not to define a unary operator to do `myVec.reduce(_ ^ _)` but to be able to xor element by element several `Vec` with the same parameter type and length.

An approach could be to work on `asBits` values and then `assignFromBits`, so just checking that the sizes in bits are the same.

Another one could be to work value-by-value, going down in the hierarchy recursively, to check that the types/sizes are correct (this way it would not be possible to xor `Vec(UInt(5 bits), 7)` with `Vec(UInt(7 bits), 5)`, which have the same size in bits but do not represent the same kind of data).

Here is what I am using for play (not type-generic or anything):

```scala
object xor {
  def apply(x: Vec[UInt]): Vec[UInt] = x
  def apply(x: Vec[UInt], y: Vec[UInt]): Vec[UInt] =
    Vec((x, y).zipped.map(_ ^ _))
  def apply(x: Vec[UInt]*): Vec[UInt] = x.reduce(xor(_, _))
}
```

So I can write `xor(vecA, vecB, vecC)` but I would like to be able to write `vecA ^ vecB ^ vecC`. So the only definition which matters is the second one, which zips and maps a xor. If `vecA ^ vecB` was possible and defined this way, then `vecvecA ^ vecvecB` would do a xor for each sub-vec, doing a xor for each value of each sub-vec, going down in the hierarchy recursively.

Dolu1990, 2022-05-25:
I just pushed a Vec Bitwise pimper, fell free to PR to complet it ^^

with it you can currently do things as : 
```scala
    val x,y = Vec(True, True)
    val z = x rawrrr y
```

numero-744, 2022-05-25:
Thanks, I didn't know the pimper idiom (I have never seen Scala code before playing with SpinalHDL), I will see what I can do :smile:

numero-744, 2022-05-28:
So the related merge request is there: https://github.com/SpinalHDL/SpinalHDL/pull/750

Now I have to understand how unit tests are done in Scala and unit-test this feature.

closed 2022-05-30 by numero-744