SpinalHDL/SpinalHDL #746: expose area that wrapped in GenerationFlags.formal.

Readon, 2022-05-23:
I want to embed some formal helper logic into the design by adding blocks with GenerationFlags.formal, the expected code are as below.
``` scala
class xxx extends Component {
    ... //the design
   GenerationFlags.formal {
        val xxxxError = CombInit(False)
        when(cond) { xxxError := True }
   }
   
   def withAsserts() {
       assert(!xxxError)
   }
   def withAssumes() {
       assume(!xxxError)
   }
}
```

1. I have tried to wrap the formal block into an Area, however in withAsserts method it could not access xxxxError.
2. if there should be an consistent name style is xxxxError preferred or errorXxxx? I think the IDE friendly should be errorXxxx, because it can quickly find what errors are defined. is there any advantages that use xxxxError?

Readon, 2022-05-23:
> I think I could let question 1 workaround by
> 
> ```
> val checker = GenerationFlags.formal {
>        new Area {
>         val xxxxError = CombInit(False)
>         when(cond) { xxxError := True }
>    }}
> ```
> 
> Then the question 2 still need to be discussed.

I personally prefer errorXxxxx,  there also can be shouldXxxxx.

Dolu1990, 2022-05-23:
@Readon 

1) Short syntax would be : 

```scala
    val checker = GenerationFlags formal new Area {

    }
```

2)

Hmmm sometime i'm more going toward : 

```scala
   GenerationFlags.formal {
        val errors = new Area{
           val xxxx = CombInit(False)
           val yyyy = CombInit(False)
        }
        when(cond) { errors.xxxx := True }
   }
```

Or : 

```scala
   GenerationFlags.formal {
        val xxxx = new Area{
           val error = CombInit(False)
           when(cond) { xxxx := True }
        }

        val yyyy = new Area{
           val error = CombInit(False)
           when(cond) { yyyy := True }
        }        
   }
```

In otherwords, structuring things more at a scala level than at a variable name.
Would that make sense ?

And if there is many error kind with simple condition, i would clearly go for val errors = new Area{ val xxxx
So the error_xxxx way.

Readon, 2022-05-23:
> 

How about the problem I met in the last post?
I have tried to put those in a standalone function and called it works, however when I  put them into the bundle it fails.
Such as outputAsserter method at line 69 in [FormalAxi4DownsizerTester](https://github.com/SpinalHDL/SpinalHDL/pull/721/files#diff-b41efbe0c288a19413ed74fdd2f64590459a313275bf45456f169cc3a03e346a), it works fine with right name. 
But after I port it to the [Axi4WriteOnly](https://github.com/SpinalHDL/SpinalHDL/blob/542d3c22b4553a27b22f20a35761ff7e01277772/lib/src/main/scala/spinal/lib/bus/amba4/axi/Axi4WriteOnly.scala#L147) file, the name system seems not work.

Readon, 2022-05-23:
> 

The short syntax seems does not work, the error message is
```
Axi4WriteOnly.scala:87:40: ';' expected but 'new' found.
[error]   val checker = GenerationFlags.formal new Area{
[error]                                        ^
[error] one error found
[error] (lib / Compile / compileIncremental) Compilation failed
```

Readon, 2022-05-23:
> How about the problem I met in the last post? I have tried to put those in a standalone function and called it works, however when I put them into the bundle it fails. Such as outputAsserter method [...]

I use an independent function to work around it. It seems that logic in bundle is not friendly to the name system, when i call a function in component the name system comes back.

Dolu1990, 2022-05-23:
GenerationFlags formal new Area { }
not 
GenerationFlags.formal new Area { } ^^

About https://github.com/SpinalHDL/SpinalHDL/blob/542d3c22b4553a27b22f20a35761ff7e01277772/lib/src/main/scala/spinal/lib/bus/amba4/axi/Axi4WriteOnly.scala#L87

Bundle are designed only to host Data, and currently on propagate names to Data : 
https://github.com/SpinalHDL/SpinalHDL/blob/dev/core/src/main/scala/spinal/core/Bundle.scala#L172

In general I think having anything else than Data in a Bundle is problematic.

What's about adding a function in the Bundle as : 
```scala
def formalContext() = new Area{
   def withAsserts() : Unit = {}
   def withCover() : Unit = {}
}
```

or something in that kind ? The current design  (https://github.com/SpinalHDL/SpinalHDL/blob/542d3c22b4553a27b22f20a35761ff7e01277772/lib/src/main/scala/spinal/lib/bus/amba4/axi/Axi4WriteOnly.scala#L147) add a tonne of hardware statement systematicaly, which seems to be quite heavy XD

Another way to handle it would be to use the userCache : https://github.com/SpinalHDL/SpinalHDL/blob/b0da3cb28d59547aefa4e7805fd4a36e8501143e/core/src/main/scala/spinal/core/Misc.scala#L138

(Currently that's only for Data, but i think we could relax that requirement to anything)
So, basicaly, withAssert would ask the cache is the formalContext logic is already created or not, and create it if not.

Readon, 2022-05-24:
> GenerationFlags formal new Area { } not GenerationFlags.formal new Area { } ^^
>
> About [...]

Good, I have adopt the formalContext way. However, if the userCache would be done, then the code can be simplified.
``` scala
val inputChecker = input.formalContext(4, 4)
inputChecker.withAssumes(maxStall)

//can be simpilified to below, however the name system might become a problem.
input.formalContext.withAssumes(maxStall)
```

Dolu1990, 2022-05-25:
signalCache can now store anything

Readon, 2022-05-25:
cool

closed 2022-05-28 by Readon