SpinalHDL/SpinalHDL #748: Does the assignment of bundle unzip the embedded Vec?

Readon, 2022-05-23:
When a bundle is [assigned as inStreams(x).payload by a record](https://github.com/SpinalHDL/SpinalHDL/blob/4d88cb390567127be094d35679dd1643259e10b1/lib/src/main/scala/spinal/lib/bus/amba4/axi/Axi4WriteOnly.scala#L223), the generated code would not assign [strbs](https://github.com/SpinalHDL/SpinalHDL/blob/4d88cb390567127be094d35679dd1643259e10b1/lib/src/main/scala/spinal/lib/bus/amba4/axi/Axi4WriteOnly.scala#L97) which is a Vec in a bundle.
``` scala
      when(awValid) {
        hist.io.inStreams(awId).payload := awRecord
        hist.io.inStreams(awId).valid := awValid
      }
```

The generated code would be
``` Verilog
    .io_inStreams_0_payload_strbs_0          (inputAssumes_checker_oRecord_strbs_0[7:0]                        ), //i
    .io_inStreams_0_payload_strbs_1          (inputAssumes_checker_oRecord_strbs_1[7:0]                        ), //i
    .io_inStreams_0_payload_strbs_2          (inputAssumes_checker_oRecord_strbs_2[7:0]                        ), //i
    .io_inStreams_0_payload_strbs_3          (inputAssumes_checker_oRecord_strbs_3[7:0]                        ), //i
```
where the oRecord is a default value of all payload, so the code of assignments seems does not work at all.

When I change the code to [assign each element of strbs separately](https://github.com/SpinalHDL/SpinalHDL/blob/4d88cb390567127be094d35679dd1643259e10b1/lib/src/main/scala/spinal/lib/bus/amba4/axi/Axi4WriteOnly.scala#L201) then the code seems right.

``` scala
      when(awValid) {
        hist.io.inStreams(awId).strbs.zip(awRecord.strbs).map { case (x, y) => x := y }
        hist.io.inStreams(awId).payload := awRecord
        hist.io.inStreams(awId).valid := awValid
      }
```

``` Verilog
    .io_inStreams_0_payload_strbs_0          (inputAssumes_checker_hist_io_inStreams_0_payload_strbs_0[7:0]    ), //i
    .io_inStreams_0_payload_strbs_1          (inputAssumes_checker_hist_io_inStreams_0_payload_strbs_1[7:0]    ), //i
    .io_inStreams_0_payload_strbs_2          (inputAssumes_checker_hist_io_inStreams_0_payload_strbs_2[7:0]    ), //i
    .io_inStreams_0_payload_strbs_3          (inputAssumes_checker_hist_io_inStreams_0_payload_strbs_3[7:0]    ), //i
```
Then the payload can be assigned from awRecord which is the only one I changed the code.

Dolu1990, 2022-05-25:
weird, could you produce a short self contained code to reproduce ?

Readon, 2022-05-25:
I will try to make one

Readon, 2022-05-28:
What I found is that the problem might occur when the assignment is indexed of Vec. 
Here is a minimal reproducable code.
``` scala
object PlayBundleAssign extends App{
  import spinal.core._
  import spinal.lib._
  case class BusCtrl() extends Bundle with IMasterSlave {
    val group = Vec(UInt(4 bits), 4)
    val data = UInt(4 bits)

    def asMaster(): Unit = {
      out(group)
      out(data)
    }
  }
  class BusDut extends Component {
    val a = slave(BusCtrl ())
    val b = Vec(master(Stream(BusCtrl())), 3)
    val c = in Bool()
    val d = in UInt(2 bits)
    val oBus = BusCtrl()
    oBus.assignFromBits(B(0, oBus.getBitsWidth bits))
    b.map(_.payload := oBus)
    b.map(_.valid := False)

    // val aBus = CombInit(oBus)
    val en = CombInit(False)
    when(c) {
      en := True
      b(d).payload := a
      b(d).valid := True
    }
  }

  SpinalConfig().generateVerilog{
    val dut = new BusDut
    dut
  }
}
```

Readon, 2022-06-07:
I have tried to trace this down, however it's still not that clear.
The above code would generate code below.
``` verilog

  assign b_0_payload_group_0 = oBus_group_0;
  assign b_0_payload_group_1 = oBus_group_1;
  assign b_0_payload_group_2 = oBus_group_2;
  assign b_0_payload_group_3 = oBus_group_3;
  always @(*) begin
    b_0_payload_data = oBus_data;
    if(c) begin
      if(_zz_2) begin    //should have b_0_payload_group_x assigned to a_group_x
        b_0_payload_data = a_data;
      end
    end
  end

```
It's obvious that the signals in group are not been assigned correctly.

PR #768 (2022-06-12): Call compositAssignFrom on Vec elements to propagate compositeAssign hook
A `compositeAssign` hook currently has no effect on the elements of a `Vec` because `Vec::assignFromImpl` calls `assignFromImpl` on its elements directly rather than via `compositAssignFrom`.  
This change aligns `Vec`'s behaviour with that of [`Bundle::assignFromImpl`](https://github.com/SpinalHDL/SpinalHDL/blob/dev/core/src/main/scala/spinal/core/Bundle.scala#L164).

I believe this fixes #748.
commit 0888436b (2022-06-12): Call Assignable::compositAssignFrom on Vec elements to propagate compositeAssign hook

Fixes #748