SpinalHDL/SpinalHDL #757: Parameterized constructor causes error

kazutoiris, 2022-06-03:
An error occurs when using the parameterized constructor.

Tested on `v1.7.0b`

```scala
package test

import spinal.core._

import scala.language.postfixOps

class TestCore(initial: Bool = False) extends Component {
  val io = new Bundle {
    val read = out(Reg(Bool())) init (False)
  }
}

object Test {
  def main(args: Array[String]) {
    SpinalVerilog(new TestCore(False))
  }
}
```

```
ERROR !
A null pointer access has been detected in the JVM.
This could happen when in your SpinalHDL description, you access an signal which is only defined further.
For instance :
  val result = Bool()
  result := a ^ b  //a and b can't be accessed there because they are only defined one line below (Software rule of execution order)
  val a,b = Bool()
          
Exception in thread "main" java.lang.NullPointerException: Cannot invoke "spinal.core.internals.ScopeStatement.append(spinal.core.internals.Statement)" because the return value of "spinal.core.DslScopeStack$.get()" is null
	at spinal.core.BaseType.assignFromImpl(BaseType.scala:227)
	at spinal.core.Assignable$class.compositAssignFrom(Trait.scala:290)
	at spinal.core.BaseType.compositAssignFrom(BaseType.scala:53)
	at spinal.core.Data$class.assignFrom(Data.scala:383)
	at spinal.core.BaseType.assignFrom(BaseType.scala:53)
	at spinal.core.internals.BoolLiteral$.apply(Expression.scala:2471)
	at spinal.core.package$.Bool(core.scala:44)
	at spinal.core.package$.False(core.scala:226)
	at naive.Test$$anonfun$main$1.apply(TestCore.scala:15)
	at naive.Test$$anonfun$main$1.apply(TestCore.scala:15)
	at spinal.core.internals.PhaseCreateComponent$$anonfun$impl$62.apply(Phase.scala:2350)
	at spinal.core.fiber.Engine$$anonfun$create$1.apply$mcV$sp(AsyncCtrl.scala:147)
	at spinal.core.fiber.AsyncThread$$anonfun$1.apply$mcV$sp(AsyncThread.scala:59)
	at spinal.core.fiber.EngineContext$$anonfun$newJvmThread$1.apply$mcV$sp(AsyncCtrl.scala:39)
	at spinal.sim.JvmThread.run(SimManager.scala:51)
```

kleinai, 2022-06-04:
Try moving the `Reg(Bool()) init (False)` out of the IO bundle. I've never done that before but it doesn't seem correct. I could be wrong.

kazutoiris, 2022-06-04:
```scala
package test

import spinal.core._

import scala.language.postfixOps

class TestCore() extends Component {
  val io = new Bundle {
    val read = out(Reg(Bool())) init (False)
  }
}

object Test {
  def main(args: Array[String]) {
    SpinalVerilog(new TestCore())
  }
}
```

What confuses me the most is that everything works fine when I use the no-parameterized constructor.

I think it's important, especially to implement some `cache`, `interface` and so on.

wswslzp, 2022-06-04:
Try `false` instead of `False`

kleinai, 2022-06-04:
I didn't catch it before. The problem is that you're passing a Spinal `Bool` in the constructor. Using `False` creates a `Bool` signal then assigns the false value to it. This pushes that assignment to the DSL stack. Because it's in the constructor the DSL stack isn't ready yet and is null.

You should avoid passing signals in as a class parameter. They won't pass the hierarchy checks.

kazutoiris, 2022-06-05:
The `Bool` behavior here should be similar to the type. I think this is useful if I want to initialize a value, especially generics `<T>`.

kleinai, 2022-06-05:
The `Bool` is not the direct problem, the `False` is because it creates a signal assignment. You're better off making an object with your input signals which then instantiates a the underlying class and wires up the signals.

See `StreamFork` for an example.
https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/Stream.scala#L852

Beyond this your component will not work as Spinal will check the hierarchy and reject it. You'll need to use an `io` bundle to pass in any signals.

kazutoiris, 2022-06-05:
Thanks for your answer, your answer is very accurate and useful. 👍

commit 54f0461c (2022-06-06): Fix #757 error message
closed 2022-06-05 by kazutoiris