SpinalHDL/SpinalHDL #758: Windows doesnt compile any code

angelsix, 2022-06-03:
Tried on 3 PCs. Works on Mac but install the setup from here https://spinalhdl.github.io/SpinalDoc-RTD/v1.5.0/SpinalHDL/Getting%20Started/getting_started.html#requirements-things-to-download-to-get-started on Windows and it never compiles.

Install Windows 10 fresh
Install IntelliJ community
Install SBT
Install Java 18 (tried 17, 8, JRE JDK, every version everywhere all the same)

Open IntelliJ and install Scala
Restart PC

Download sample projects (also have working commercial products I compile on my Mac which works and tested the same).
Run the sample code to compile and always get this:

```
C:\Users\conta\.jdks\openjdk-18.0.1.1\bin\java.exe "-javaagent:C:\Program Files\JetBrains\IntelliJ IDEA Community Edition 2022.1.2\lib\idea_rt.jar=59204:C:\Program Files\JetBrains\IntelliJ IDEA Community Edition 2022.1.2\bin" -Dfile.encoding=UTF-8 -classpath C:\Users\conta\Desktop\SpinalTemplateSbt-master\target\scala-2.11\classes;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\oshi\oshi-core\5.2.0\oshi-core-5.2.0.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\scopt\scopt_2.11\3.7.1\scopt_2.11-3.7.1.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\spinalhdl\spinalhdl-core_2.11\1.7.0a\spinalhdl-core_2.11-1.7.0a.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\spinalhdl\spinalhdl-idsl-payload_2.11\1.7.0a\spinalhdl-idsl-payload_2.11-1.7.0a.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\spinalhdl\spinalhdl-idsl-plugin_2.11\1.7.0a\spinalhdl-idsl-plugin_2.11-1.7.0a.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\spinalhdl\spinalhdl-lib_2.11\1.7.0a\spinalhdl-lib_2.11-1.7.0a.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\spinalhdl\spinalhdl-sim_2.11\1.7.0a\spinalhdl-sim_2.11-1.7.0a.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\lihaoyi\sourcecode_2.11\0.2.7\sourcecode_2.11-0.2.7.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\commons-io\commons-io\2.4\commons-io-2.4.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\net\java\dev\jna\jna-platform\5.5.0\jna-platform-5.5.0.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\net\java\dev\jna\jna\5.5.0\jna-5.5.0.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\net\openhft\affinity\3.21ea1.1\affinity-3.21ea1.1.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\jetbrains\annotations\19.0.0\annotations-19.0.0.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\scala-lang\modules\scala-parser-combinators_2.11\1.0.4\scala-parser-combinators_2.11-1.0.4.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\scala-lang\modules\scala-xml_2.11\1.0.5\scala-xml_2.11-1.0.5.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\scala-lang\scala-compiler\2.11.12\scala-compiler-2.11.12.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\scala-lang\scala-library\2.11.12\scala-library-2.11.12.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\scala-lang\scala-reflect\2.11.12\scala-reflect-2.11.12.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\slf4j\slf4j-api\1.7.30\slf4j-api-1.7.30.jar;C:\Users\conta\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\slf4j\slf4j-simple\1.7.25\slf4j-simple-1.7.25.jar mylib.MyTopLevelVerilog
[Runtime] SpinalHDL v1.7.0a    git head : 150a9b9067020722818dfb17df4a23ac712a7af8
[Runtime] JVM max memory : 16304.0MiB
[Runtime] Current date : 2022.06.03 10:20:32
[Progress] at 0.000 : Elaborate components

**********************************************************************************************
[Warning] Elaboration failed (0 error).
          Spinal will restart with scala trace to help you to find the problem.
**********************************************************************************************

[Progress] at 0.725 : Elaborate components
Exception in thread "main" java.lang.IllegalArgumentException: Argument value 0xfffffffff exceeds native capacity (4 bytes) mask=0xffffffff00000000
	at com.sun.jna.IntegerType.setValue(IntegerType.java:112)
	at com.sun.jna.IntegerType.<init>(IntegerType.java:69)
	at com.sun.jna.platform.win32.WinDef$DWORD.<init>(WinDef.java:144)
	at net.openhft.affinity.impl.WindowsJNAAffinity.setAffinity(WindowsJNAAffinity.java:94)
	at net.openhft.affinity.Affinity.setAffinity(Affinity.java:159)
	at spinal.core.fiber.EngineContext.start(AsyncCtrl.scala:96)
	at spinal.core.fiber.Engine$.create(AsyncCtrl.scala:150)
	at spinal.core.internals.PhaseCreateComponent.impl(Phase.scala:2345)
	at spinal.core.internals.PhaseContext.doPhase(Phase.scala:181)
	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$129.apply(Phase.scala:2650)
	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$129.apply(Phase.scala:2648)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2648)
	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2584)
	at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:69)
	at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2584)
	at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2579)
	at spinal.core.Spinal$.apply(Spinal.scala:383)
	at spinal.core.SpinalConfig.generate(Spinal.scala:163)
	at spinal.core.SpinalVerilog$.apply(Spinal.scala:402)
	at mylib.MyTopLevelVerilog$.main(MyTopLevel.scala:47)
	at mylib.MyTopLevelVerilog.main(MyTopLevel.scala)
```

angelsix, 2022-06-03:
Also FYI installing Ubuntu subsystem on this same machine and doing the following builds fine:

```
sudo apt-get install openjdk-8-jdk
sudo apt-get install scala
echo "deb https://repo.scala-sbt.org/scalasbt/debian all main" | sudo tee /etc/apt/sources.list.d/sbt.list
echo "deb https://repo.scala-sbt.org/scalasbt/debian /" | sudo tee /etc/apt/sources.list.d/sbt_old.list
curl -sL "https://keyserver.ubuntu.com/pks/lookup?op=get&search=0x2EE0EA64E40A89B84B2DF73499E82A75642AC823" | sudo apt-key add
sudo apt-get update
sudo apt-get install sbt
git clone https://github.com/SpinalHDL/SpinalTemplateSbt.git SpinalTemplateSbt
cd SpinalTemplateSbt
sbt run   # select "mylib.MyTopLevelVhdl" in the menu
ls MyTopLevel.vhd
```

Readon, 2022-06-03:
I have succeeded in eleborate design on MSys2 with verilator.

Dolu1990, 2022-06-03:
Ahhh damned, net.openhft.affinity again, a nest of compatibility issues :/

kleinai, 2022-06-04:
I have been running purely on Windows with msys2 with every (open source) simulator. I'm using OpenJDK 13.

wswslzp, 2022-06-04:
I recommend using WSL to install SpinalHDL environment.

angelsix, 2022-06-04:
I have it working on WSL but thats a lot more steps and its also a lot longer to prepare the environment every time for quick development, having to navigate in bash to the windows folders and run a command every time, vs using the IntelliJ IDEA to just quickly build.

Following the exact install instructions and links on the official site do not work for Windows 10 or 11 right now, at all.

wswslzp, 2022-06-04:
You can also install IntelliJ IDEA on WSL, then run your IDEA just in WSL ubuntu to enable quick build.

> Following the exact install instructions and links on the official site do not work for Windows 10 or 11 right now, at all.

Indeed native Windows environment sucks, but WSL/WSL2 on Win10/11 is good enough to run Scala.

angelsix, 2022-06-04:
Yes good point. Perhaps update the documentation to correct the fact it does not work for Windows at the moment and to use WSL with a basic guide on how to do that for others also

Dolu1990, 2022-06-06:
@angelsix I pushed a new version in dev which try / catch the openhft affinity calls, so it may be fine on your side too now. 
Could you try with dev branch ?

commit 3791d1cb (2022-06-06): try / catch on all CPU affinity changes to fix #758 while things sort themself with new versions
closed 2022-06-06 by Dolu1990