SpinalHDL/SpinalHDL #761: Is there option to not merge syncProcess

dxsm, 2022-06-05:
Hi @Dolu1990 , I notice asyncProcess has the option to choose whether to merge to a single process. Why syncProcess does not have similar option? Different functions should not merge together when I use different `when` scope.

kleinai, 2022-06-12:
Async processes are tracked by tree statements, switches and when statements. Sync processes are tracked by scope statements. I don't see any obvious reason why this couldn't be changed to have similar behavior as async processes.

kleinai, 2022-06-12:
Putting a `RegNext` in a `when` statement makes a different clocked process. Do you have an example of where you're seeing the sync process merge?

Dolu1990, 2022-06-13:
Hi @dxsm 

There is a few reason for the current states of the backend : 
- Sync stuff are regrouped in a single always block in order to reduce netlist size / reduce simulation overhead (potentialy)
- Async stuff have each their own always block in order to avoid the verilog simulators enterring dead loops by triggered by cross dependent always blocks (no hardware combinatorial loop, but simulation always blocks depending on each others creating endless loop)

So, the reasons are currently technical.

About having different syncProcess for each functions, i would say, what is a function ?

dxsm, 2022-06-29:
```scala
class test1 extends Module {
  val a,b, c, d, e = UInt(10 bits) setAsReg()
  val en1, en2 = Bool()

  a := 6
  b := 7

  when(en1) {
    c := 4
    d := 5
  }

  when(en2) {
    e := 4
  } elsewhen(en1) {
    e := 5
  }
}
```
a/b、c/d、e may have different sync group. All sync process like switch FSM、different when scope grouped makes RTL inconsistent with scala which make me take more time to find it's location when add signal to verdi.
Also netlist size reduction is limited, only one line for each register.  So expect merge syncProcess to be optional.