SpinalHDL/SpinalHDL #769: Supporting SpinalHDL

Dolu1990, 2022-06-08:
As a way to scale up SpinalHDL we are looking at : 

- Companies sponsorship as a way to provide funding to the SpinalHDL maintenance
- Open a https://opencollective.com/ or https://liberapay.com/ (or something else) in order to connect and mannage donnations
- Maintainers to keep up with the increasing flow of question / issues / pull requests

Let us know if you have any comment / opinon / interest in any of those topic ^^

Regards
Charles

Readon, 2022-06-10:
It's really important to have some support for long term supporting.
How about creating an open collective organization first?

Dolu1990, 2022-06-10:
Done ^^ 
https://opencollective.com/spinalhdl
https://opencollective.com/spinalhdl/conversations/projects-funding-4qj7kkq8

Readon, 2022-06-10:
> Done ^^ https://opencollective.com/spinalhdl https://opencollective.com/spinalhdl/conversations/projects-funding-4qj7kkq8

I think, it's best to add the sponsor link to the homepage of both the project and documentation.

Readon, 2022-06-10:
Some other project would distribute their roadmap or plan for new features and related budget, the task would be start when the first money collection milestone target achieve. That might be an option.

kleinai, 2022-06-10:
If we want we could start making tickets for big milestones. Maybe we're a little ad hoc for that right now

Readon, 2022-06-10:
> If we want we could start making tickets for big milestones. Maybe we're a little ad hoc for that right now

For example, I am very interested to bring a good pipeline mechanism into the library, such as it is used in NaxRiscV previously.
If the plan is distributed and call for help, I would do something I could.

Althrough, Dolu1990 have already processing on this, XD.

Dolu1990, 2022-06-10:
> I think, it's best to add the sponsor link to the homepage of both the project and documentation.

I was waiting a bit to see if that's seems ok for others, but o well ^^
Added :)

> pipeline / Althrough, Dolu1990 have already processing on this, XD.

So, i merged it into SpinalHDL dev, it is far from perfect. If you want we can do a call for me to demo it ^^

> roadmap 

This would be more like a list of wished features ? Roadmap sound like static / linear

kleinai, 2022-06-10:
> > roadmap
> 
> This would be more like a list of wished features ? Roadmap sound like static / linear

Roadmap might not be as appropriate here as this isn't a product with defined features to build out.  Although it would be useful if there's a list of wants to work for.  Pipelining is cool and could use collaboration.  I'd personally like to finish more of RegIf as it seems promising.

Readon, 2022-06-11:
I don't know if the procedure cost of 3.08/20 is a normal rate? I my opinion, it is relative high.

Dolu1990, 2022-06-11:
> I don't know if the procedure cost of 3.08/20 is a normal rate? I my opinion, it is relative high.

Yes, i'm a bit surprised it is 15%,  The collective i created is : 
https://opencollective.com/create/opensource

and the 5 % extra probably come from : 
"5% fee on each incoming contribution made via credit card"
https://docs.opencollective.foundation/how-it-works/fees

seems they charge that much to have a "fiscal host" :
https://opencollective.com/fiscal-hosting

Apparently our "fiscal host" is : 
https://opencollective.com/opensource

Question now is what is the real fee of transfering money to a bank acounts, and what realy bring a "fiscal host" on the table.

Readon, 2022-06-11:
how about [OSSi](https://www.fossi-foundation.org/)?

Dijky, 2022-06-11:
> Question now is what is the real fee of transfering money to a bank acounts,

OSC takes a 10% fee on each donation for administration.  
The project also pays all payment processor fees (e.g. to PayPal) both from the donor to the project, and from the project to you.

Stripe in particular seems to take 2.9% + $0.30 for credit card transactions. Plus 1% for non-US cards, plus 1% for currency exchange to USD.  
For @Readon's donation, $1.08 of $20 went to Stripe. That is exactly 3.9% \* $20 + $0.30 (non-US card it seems).  
$2.00 went to OSC (10%).

What's shown on OpenCollective.com is the amount after administration fee (for OSC) and payment processor fee *to* the project. From what I can tell, the payment processor fee from the project to you is deduced from that when you request a payout.

The payment processor fees vary on both ends of the chain by the chosen processor (Stripe, PayPal etc.), amount per transaction (because there are fixed fees and percentages), country and currency of origin/destination. They are a pain to figure out.

> and what realy bring a "fiscal host" on the table.

To accept money, the project needs some legal entity.  
You could accept money personally, but then that may count as your personal income and may end up requiring a legal entity anyways (depending on your local laws, e.g. some self-employment registration). It gets more complicated if you want to distribute the money to other contributors.  
So you need something like a non-profit company or charity that owns a bank account and receives the money. But that requires administrative work as well, such as registering, proper accounting, tax reportsand forms etc.

OSC offers to be that non-profit company, run the bank account and do the financial/tax administrative work on behalf of the project. Contributors can then request payouts to cover bills they paid for the project ("reimbursement", e.g. to buy a testing device) or effort they put in ("invoice").  
That's what they call a "fiscal host".

Now, you don't need to do the administration for the whole project, although if you personally get a payout, you may still need to declare that payout as personal freelance or private business income, or however it works in your country.

numero-744, 2022-11-22:
For now, for PR reviews tend to `git blame` modified files to try to find active people who have modified the same files before and **@ping** them.

About scaling up for the long term, how decentralized is ownership? For instance, @Dolu1990 is really busy these times, and if one day he has an accident (I hope he won't!), what would be the consequences for the project?

kleinai, 2022-11-22:
> About scaling up for the long term, how decentralized is ownership? For instance, @Dolu1990 is really busy these times, and if one day he has an accident (I hope he won't!), what would be the consequences for the project?

Absolute worse case, the largest public fork of the project carries on development. It's all open source after all!

Dolu1990, 2022-11-22:
My personnal favorite is death by chocolate <3

But for SpinalHDL, i would say, that would not be a big deal, the SpinalHDL core isn't rocket science at all. Today, most of the movement is done on the SpinalHDL lib.

The main thing would be about to release new versions of SpinalHDL ^^

numero-744, 2022-11-22:
> death by chocolate <3

XD

> Absolute worse case, the largest public fork of the project carries on development. It's all open source after all!

> But for SpinalHDL, i would say, that would not be a big deal, the SpinalHDL core isn't rocket science at all. Today, most of the movement is done on the SpinalHDL lib.

IMHO you both adopt a narrow, pure technical point of view, not considering the project as a whole here… This is not only about code. The day @Dolu1990 stops working on it for professional or other reasons, the project just dies, because it loses all its structure (opencollective, release keys, repository ownership, funding, email address, contributors, users, relations, name, etc.).

Scaling up also means building an entity so that SpinalHDL can become "the SpinalHDL project, by SpinalHDL team/association/whatever and initiated by @Dolu1990" and not "the project of @Dolu1990 with help from his friends". And by "building an entity" I mean having several people managing it, not just knowing that @Dolu1990 has opened an account named SpinalHDL on opencollective. Scale this project up!

Having @Dolu1990 as a leader is a good thing, but only relying on a single person is not a stable situation on the long term.

Dolu1990, 2022-11-23:
> The day @Dolu1990 stops working on it for professional or other reasons, the project just dies

Hmm i don't think, there is now quite many people with the repo rights and contribute to it, the core of the project would survive for sure ^^
The two critical things are : release keys, repository ownership, the rest depends mostly on the survivors XD
/joke Worst worst worst case, fork the repo in a new organization named "PhoenixHDL" or "SpinalHDLForever" <3<3 /joke 

> Scaling up also means building an entity so that SpinalHDL can become "the SpinalHDL project, by SpinalHDL team/association/whatever and initiated by @Dolu1990" and not "the project of @Dolu1990 [...]

I have fondamentaly nothing against it, the issue are : 
- There is close to no money in SpinalHDL
- People which use SpinalHDL much are generaly already quite busy with their work

So currently, nearly everything rely on the goodwill of peoples not being too much stressed by work to contribute back to SpinalHDL. 

Since maybe 6 months there was already quite more activity from "not me" in the contribution, which is great :D Quite "organic" grow XD

So the question is how do you find the poeple to scale up ?

andreasWallner, 2022-11-27:
> So the question is how do you find the poeple to scale up ?

I'm not sure there is a solution for that in open source. From other projects it really just seems about people appreciating what the original author put in, starting to do small contributions (e.g. to see how the culture is) and maybe go from there...