SpinalHDL/SpinalHDL #773: getAheadValue could not get correct bits width while partial assignment is used.

Readon, 2022-06-10:
getAheadValue seems not compatible with partial assignment.
I modify the tester in CoreMiscTester, it will fail the elaborating process with width mismatch errors.
pay attention to the f(0, 8bits) and f(8, 8 bits).

``` scala
  test("reg_nextValue"){
    SimConfig.compile(new Component{
      val conds = in(Vec.fill(8)(Bool()))
      val a, b, c, d, e = out(Reg(UInt(8 bits)) init(0))
      val f = out(Reg(UInt(16 bits)) init(0))  //here is what I added.

      when(conds(0)){
        a := 1
        when(conds(1)){
          a := 2
          b := 11
        } otherwise {
          a := 3
          c := 21
        }
        f(0, 8 bits) := a  //here is the parial assignment.
      }
      when(conds(2)){
        a := 4
        b := 12
        c := 22
        d := 31
        f(8, 8 bits) := a
      }

      val x = out(a.getAheadValue)
      val y = out(a.getAheadValue)
      val z = out(b.getAheadValue)
      val w = out(f.getAheadValue)

      when(d.getAheadValue() === 0){
        e := 1
      }
    }).doSim(seed = 42){dut =>
      var an,bn,cn,a,b,c = 0
      dut.conds.foreach(_ #= false)
      dut.clockDomain.forkStimulus(10)
      dut.clockDomain.waitSampling()

      for(i <- 0 until 1000){
        if(dut.conds(0).toBoolean){
          an = 1
          if(dut.conds(1).toBoolean){
            an = 2
            bn = 11
          } else {
            an = 3
            cn = 21
          }
        }
        if(dut.conds(2).toBoolean){
          an = 4
          bn = 12
          cn = 22
        }
        assert(dut.x.toInt == an)
        assert(dut.y.toInt == an)
        assert(dut.z.toInt == bn)
        assert(dut.a.toInt == a)
        assert(dut.b.toInt == b)
        assert(dut.c.toInt == c)
        a = an
        b = bn
        c = cn
        dut.clockDomain.waitSampling()
        dut.conds.foreach(_.randomize())
      }
    }
  }
```

Here is the error message.

``` log
WIDTH MISMATCH (16 bits <- 8 bits) on (toplevel/w : out UInt[16 bits]) := (toplevel/a : out UInt[8 bits]) at  
[info]     spinal.tester.scalatest.CoreMiscTester$$anonfun$2$$anonfun$apply$mcV$sp$5$$anon$3$$anonfun$3.apply$mcV$sp(CoreMiscTester.scala:47)                                                                                              
[info]     spinal.tester.scalatest.CoreMiscTester$$anonfun$2$$anonfun$apply$mcV$sp$5$$anon$3.<init>(CoreMiscTester.scala:38)                                                                                                               
[info]     spinal.tester.scalatest.CoreMiscTester$$anonfun$2$$anonfun$apply$mcV$sp$5.apply(CoreMiscTester.scala:33)  
[info]     spinal.tester.scalatest.CoreMiscTester$$anonfun$2$$anonfun$apply$mcV$sp$5.apply(CoreMiscTester.scala:33)  
[info]     spinal.sim.JvmThread.run(SimManager.scala:51)
```

Readon, 2022-06-10:
If I create a wire to separate the partial assignment and the getAheadValue as below.

``` scala
      val g = UInt(16 bits)  //this is the intermediate wire.
      val f = RegNext(g) init(0)
      g := f

      when(conds(0)){
        a := 1
        when(conds(1)){
          a := 2
          b := 11
        } otherwise {
          a := 3
          c := 21
        }
        g(0, 8 bits) := a
      }
      when(conds(2)){
        a := 4
        b := 12
        c := 22
        d := 31
        g(8, 8 bits) := a
      }
```
then an error of assignment overlap reported
``` log
ASSIGNMENT OVERLAP completely the previous one of (toplevel/w : out UInt[16 bits])                            
[info]     spinal.tester.scalatest.CoreMiscTester$$anonfun$2$$anonfun$apply$mcV$sp$5$$anon$3.<init>(CoreMiscTester.scala:37)                                                                                                               
[info]     spinal.tester.scalatest.CoreMiscTester$$anonfun$2$$anonfun$apply$mcV$sp$5.apply(CoreMiscTester.scala:33)  
[info]     spinal.tester.scalatest.CoreMiscTester$$anonfun$2$$anonfun$apply$mcV$sp$5.apply(CoreMiscTester.scala:33)  
[info]     spinal.sim.JvmThread.run(SimManager.scala:51)   
```

Dolu1990, 2022-06-10:
I pushed a fix, not be all good now.

Readon, 2022-06-10:
> I pushed a fix, not be all good now.

nice, it works.

closed 2022-06-10 by Readon