SpinalHDL/SpinalHDL #775: Question about simulation on design with memory initialization on windows

DoubiZhizun, 2022-06-13:
Hi
I want to simulate a design with memory initialization on windows, it should generate a bin file with initialization data and use $readmemb in design to read it, but when I start simulation, an error happen:
```
%Error: C:/Users/14112/Desktop/ysyx/tmp/job_1/cpu.v:1463:15: Unknown escape sequence: \U
 1463 |     $readmemb("C:\Users\14112\Desktop\ysyx\.\simWorkspace\cpu\rtl/cpu.v_toplevel_instMem_1__zz_1.bin",_zz_1);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
It seems that verilator seen '\U' as escape sequence, but it just a file separator followed by 'U'.
Is this a bug? Maybe it should generate '\\\\' or '/' as file separator?
And what should I do if I want to simulate my design?

Dolu1990, 2022-06-13:
Hi,

Which version of SpinalHDL are you using ?

If a remember well there was a few related changes / fixes recently.

DoubiZhizun, 2022-06-13:
Hi
It seems that it's been too long since my last update, my version is 1.6.4.
After I update to 1.7.0 this problem disappeared.
Thanks for your  answer.

closed 2022-06-13 by DoubiZhizun