SpinalHDL/SpinalHDL #778: load repeat data when use loadBinary

aiit-linry, 2022-06-14:
https://github.com/SpinalHDL/SpinalHDL/blob/703a281be9fc1e918e1f49431e3e22591b561c2c/lib/src/main/scala/spinal/lib/bus/amba4/axi/sim/AxiMemorySim.scala#L142

When the length over one page, data should be cut to slice. But use drop() in this line is error, data wont drop itself. This function should like this:

var data_temp = data
...
...
for(xx){
..
val bytesWritten = page.writeArray(offset, data_temp)
data_temp = data_temp.drop(bytesWritten)
..
}

aiit-linry, 2022-06-14:
scala version: 2.11.12

hanm2019, 2022-06-22:
I found the same problem, and pull a merge request #784

aiit-linry, 2022-08-11:
> @aiit-linry You can test again on `dev` branch

test pass in 1.7.2, thx

closed 2022-08-11 by aiit-linry