SpinalHDL/SpinalHDL #779: Feature: * operation for Bits

dnltz, 2022-06-14:
Hi,

I just needed to concat a bits data type n-times and tried the * operation, which is implemented for Strings:

```
val foo = B"0001"
val bar := sync * 5
```

Well, this operation is not implemented and I actually really like the *-concat method. Do we want to add this operation to bits? Are there other simple ways with the same result, which are missing in the doc?

Dolu1990, 2022-06-14:
lol ^^

Why not :)

B(miaou)*5
5*B(miaou) could be usefull / natural

dnltz, 2022-06-15:
I'm unsure if `5 * B(...` works because it would require overloading the *-method of Int. I'll create a PR soon :)

Dolu1990, 2022-06-15:
> I'm unsure if 5 * B(... works because it would require overloading the *-method of Int. I'll create a PR soon :)

Tricks, yes yes, tricks ! 
https://github.com/SpinalHDL/SpinalHDL/commit/5d0862465ff25cc91c26751110e2ece3d6a2590a#diff-8e04bec1093f9c0adf1554be67efcb103762b35bf677fe76fbb51fd432e4f827R296

tanhongze, 2022-06-15:
I worry about that it could be confusing if `Int * Int`, `Int * UInt` and `Int * Bits` occur at the same time.
Maybe using an operator like `*^`  `*:` could be better than simply using `*`?

Dolu1990, 2022-06-15:
> your patch only covers Int * Bits, right?

Yes

> I would suggest #*

Ok for me ^^

Dolu1990, 2022-06-15:
I would add, that #* could then be applied to more general types, so, it could be implemented for Data directly, not only Bits. which has some adventages

wswslzp, 2022-06-17:
I have pushed a feature that repeats n times of a bit string. You can check out #760

closed 2022-06-20 by Dolu1990