SpinalHDL/SpinalHDL #780: Feature: Add TracingOn Tag?

tanhongze, 2022-06-15:
Sometimes, I just want tracing a few sub-modules.
So, I want to put a `TracingOff` in top module and `TracingOn`s in my cared modules to reopen tracing.
The expected verilog would be generated like:
```
module top();
/*verilator tracing_off*/
// ignoring contents
/*verilator tracing_on*/
cared_module m(...);
/*verilator tracing_off*/
// other ignoring contents
/*verilator tracing_on*/
endmodule
```

kleinai, 2022-06-16:
Spinal doesn't yet have a system to selectively trace simulation signals.

numero-744, 2022-10-03:
What is the benefit of tracing only a few sub-modules? Would it bring performance improvements?

Dolu1990, 2022-10-04:
It reduce trace size / simulation speed penality.

Dolu1990, 2022-10-04:
Added as : 

```scala
    SimConfig.withFstWave.doSim(new Component{
      val id = U(0x1)
      val subA = new Component{
        val id = U(0x10)
        val subB = new Component{
          val id = U(0x100)
        }

        traceDisable()
      }
      val subC = new Component{
        val id = U(0x20)
        val subD = new Component{
          val id = U(0x200)

        }
      }
    }){ dut =>
        sleep(100)
    }
```

Note traceDisable will apply recursively by default on all sub component already existing.
traceDisable(false) will only apply to the  component on which it is called

closed 2022-10-20 by numero-744