SpinalHDL/SpinalHDL #790: val ... = Counter(...).value.dontSimplifyIt() don't work

wmj0607, 2022-07-06:
Example(similar to lib.....RiscvCore.performanceCounters I believe):
```scala
object Test {
  def main(args: Array[String]) {SpinalConfig(mode = Verilog, removePruned = true).generate{new Component{
    val i = in(Bool())
    val cnt = Counter(8.bits, i).value.dontSimplifyIt()
  }}}
}
```
will generates an empty module (same even if set removePruned = false)

However if wrote
```scala
    val cnt = Counter(8.bits, i)
    cnt.value.dontSimplifyIt()
```
will generates the counter properly.

Dolu1990, 2022-07-06:
Hi,
I would say it isn't realy an issue of dontSimplifyIt.

dontSimplifyIt is mostly legacy from a time where SpinalHDL was removing from the netlist every signals which was not usefull for the behaviour of the toplevel outputs.

About your example, thing thing is that  val cnt = Counter(8.bits, i).value.dontSimplifyIt() will not provide a name for anything : 
- Counter will not be named as cnt will try to name the "value"
- "value" will not be named by "cnt" as it is a child of Counter already.

So nothing is named and in that case SpinalHDL will consider that it can remove things, even if there is the dontSimplifyIt.

Note that Counter is implicitly converted into its value so you don't need to explicitly do myCounter.value when you want to use it.

wmj0607, 2022-07-06:
Thanks, got why that case doesn't work.

 I also noticed that `val cnt = Counter(8.bits, i).dontSimplifyIt()` also doesn't work, where I suppose the Counter will be named by cnt. 
Then I assigned it to some output and it turns out the Counter was named \_zz\_.
I guess the real problem is that dontSimplifyIt prevents SpinalHDL from naming the Counter? 

> Note that Counter is implicitly converted into its value so you don't need to explicitly do myCounter.value when you want to use it.

This is actually a imitate to `lib/src/main/scala/spinal/lib/cpu/riscv/impl/RiscvCore.scala`, where it wrotes
```scala
  val performanceCounters = new Area{
    val decode_pcLoad = Counter(1<<30,decode.pcLoad.valid).value.dontSimplifyIt()
    val execute1_pcLoad = Counter(1<<30,execute1.pcLoad.valid).value.dontSimplifyIt()

    val decode_halt = Counter(1<<30,decode.halt && decode.inInst.valid).value.dontSimplifyIt()
    val execute0_halt = Counter(1<<30,execute0.halt && execute0.inInst.valid).value.dontSimplifyIt()
    val execute1_halt = Counter(1<<30,execute1.halt && execute1.inInst.valid).value.dontSimplifyIt()
    val writeBack_halt = Counter(1<<30,writeBack.halt && writeBack.inInst.valid).value.dontSimplifyIt()
  }
```

Dolu1990, 2022-07-14:
> val cnt = Counter(8.bits, i).dontSimplifyIt() also doesn't work

It is because dontSimplifyIt do not exist in the Counter class (it is an Area), so what that line do, is implicitly convert the Counter into its .value, and then apply the dontSimplifyIt on it, then return value. It is exactly as : 
val cnt = Counter(8.bits, i).value.dontSimplifyIt() 
So, it is normal.

One question is why do you want to use the dontSimplifyIt ? I guess with recent version of SpinalHDL, it is mostly a useless feature (dontSimplifyIt).
What's about just "val decode_pcLoad = Counter(1<<30,decode.pcLoad.valid)" ?

wmj0607, 2022-07-14:
> One question is why do you want to use the dontSimplifyIt?

What I want is to have some counters for bebug and performance measure, which is "unused" and will be pruned.

Despite all those SpinalHDL warnings, remove dontSimplifyIt and set removepruned = false seems fine to me.
What I do now is name the Counter and then call dontSimplifyIt on value though.

BTW: since keep() is removed, the document (Structuring/Component and hierarchy/Pruned signals) is out dated.

closed 2022-07-14 by wmj0607

Dolu1990, 2022-07-18:
Ahhh, i'm not using much  removepruned = true , but with it enable, you are right ^^

> What I do now is name the Counter and then call dontSimplifyIt on value though. 

Yes, that's should be fine.

I just updated the doc
thanks ^^