SpinalHDL/SpinalHDL #796: 「BUG?」When Merging RTL

dreamflyings, 2022-07-21:
1. If spinalsystemverilog is used, Verilog is generated after mergertlsource instead of SystemVerilog
2. If spinalconfig is configured as onefilepercomponent, and only report blackboxesSourcesPaths. Add, but without mergertlsource, the files in blackbox addrtlpath will not be copied.
THX！

Dolu1990, 2022-07-24:
Hi,
Do you have a simple self contained example to reproduce ?

dreamflyings, 2022-07-25:
> Hi, Do you have a simple self contained example to reproduce ?
Thank you!
Examples of recurrence are as follows:
```scala
import spinal.core._

object MergeRTL0 extends App{
  val report = SpinalSystemVerilog{
    new Module{
      val a = in UInt(8 bits)
      val b = in UInt(8 bits)
      val c = out UInt(8 bits)
      val subMod = new BlackBox {
        setDefinitionName("Sub")
        noIoPrefix()
        val a = in UInt(8 bits)
        val b = in UInt(8 bits)
        val c = out UInt(8 bits)
        setInlineVerilog(
          s"""module Sub(
              | input [7:0] a,
              | input [7:0] b,
              | output[7:0] c
              |);
              | assign c = a + b + (a[1] ^ b[1]);
             |endmodule
             |""".stripMargin)
      }
      subMod.a := a
      subMod.b := b
      c := subMod.c
    }.setDefinitionName("Top")
  }
}

object MergeRTL1 extends App{ // todo Problem 1
  val report = SpinalSystemVerilog{
    class SubMod extends BlackBox {
      setDefinitionName("Sub")
      noIoPrefix()
      val a = in UInt(8 bits)
      val b = in UInt(8 bits)
      val c = out UInt(8 bits)
      addRTLPath("./Sub.sv")
    }
    new Module{
      val a = in UInt(8 bits)
      val b = in UInt(8 bits)
      val c = out UInt(8 bits)
      val subMod = new SubMod
      subMod.a := a
      subMod.b := b
      c := subMod.c
    }.setDefinitionName("Top")
  }
  report.blackboxesSourcesPaths.add("./Top.sv")
  report.mergeRTLSource("Top1")
}

object MergeRTL2 extends App{// todo Problem 1
  val report = SpinalSystemVerilog{
    new Module{
      val a = in UInt(8 bits)
      val b = in UInt(8 bits)
      val c = out UInt(8 bits)
      val subMod = new BlackBox {
        setDefinitionName("Sub")
        noIoPrefix()
        val a = in UInt(8 bits)
        val b = in UInt(8 bits)
        val c = out UInt(8 bits)
        setInlineVerilog(
          s"""module Sub(
             | input [7:0] a,
             | input [7:0] b,
             | output[7:0] c
             |);
             | assign c = a + b + (a[1] ^ b[1]);
             |endmodule
             |""".stripMargin)
      }
      subMod.a := a
      subMod.b := b
      c := subMod.c
    }.setDefinitionName("Top")
  }
  report.blackboxesSourcesPaths.add("./Top.sv")
  report.mergeRTLSource("Top2")
}

object MergeRTL3 extends App{ // todo problem 2
  val report = SpinalSystemVerilog(SpinalConfig(oneFilePerComponent = true)){
    new Module{
      val a = in UInt(8 bits)
      val b = in UInt(8 bits)
      val c = out UInt(8 bits)
      val subMod = new BlackBox {
        setDefinitionName("Sub")
        noIoPrefix()
        val a = in UInt(8 bits)
        val b = in UInt(8 bits)
        val c = out UInt(8 bits)
//        setInlineVerilog(
//          s"""module Sub(
//             | input [7:0] a,
//             | input [7:0] b,
//             | output[7:0] c
//             |);
//             | assign c = a + b + (a[1] ^ b[1]);
//             |endmodule
//             |""".stripMargin)
        addRTLPath("./Sub.sv")
      }
      subMod.a := a
      subMod.b := b
      c := subMod.c
    }.setDefinitionName("Top")
  }
  //  report.blackboxesSourcesPaths.add("./Top.sv")
}

object MergeRTL4 extends App{ // todo problem 2
  val report = SpinalSystemVerilog(SpinalConfig(oneFilePerComponent = true)){
    new Module{
      val a = in UInt(8 bits)
      val b = in UInt(8 bits)
      val c = out UInt(8 bits)
      val subMod = new BlackBox {
        setDefinitionName("Sub")
        noIoPrefix()
        val a = in UInt(8 bits)
        val b = in UInt(8 bits)
        val c = out UInt(8 bits)
//                setInlineVerilog(
//                  s"""module Sub(
//                     | input [7:0] a,
//                     | input [7:0] b,
//                     | output[7:0] c
//                     |);
//                     | assign c = a + b + (a[1] ^ b[1]);
//                     |endmodule
//                     |""".stripMargin)
        addRTLPath("./Sub.sv")
      }
      subMod.a := a
      subMod.b := b
      c := subMod.c
    }.setDefinitionName("Top")
  }
    report.blackboxesSourcesPaths.add("./Top.sv")
//  report.mergeRTLSource("Top3")
}

```
```verilog
`timescale 1ns/1ps 
module Sub(
 input [7:0] a,
 input [7:0] b,
 output[7:0] c
);
 assign c = a + b + (a[1] ^ b[1]);
endmodule
```

dreamflyings, 2022-07-25:
> Hi, Do you have a simple self contained example to reproduce ?

By the way, how can I use VCS simulator to simulate according to SV instead of Verilog. This function seems to be very important (because there are many SystemVerilog for UVM verification). Please pay attention to this.
Thank you very much!

kleinai, 2022-08-02:
Check out #810. I fixed the merged file naming and add blackbox RTL paths to the resulting lst file.

closed 2022-09-20 by Dolu1990