SpinalHDL/SpinalHDL #797: It's best to put SpinalFormalFunSuite.scala into lib or sim to help project test with formal.

Readon, 2022-07-22:
The existing SpinalFormalFunSuite is a wrap of FunSuite for formal verification.
However, this can only be used to verify SpinalHDL itself.
Move SpinalFormalFunSuite to lib or sim subproject to provide the accessibility of projects using Spinal is necessary.

The problem is where to put this file. 
It seems that lib subproject is not suitable for test related code.
However, sim subproject is also not very fit for that, because it's mainly target to simulation.

Dolu1990, 2022-07-22:
Ahh why not :)
spinal.lib.formal i would say is a good place

Readon, 2022-07-22:
> Ahh why not :) spinal.lib.formal i would say is a good place

That would lead to the dependency of org.scalatest.

PR #802 (2022-07-23): move SpinalFormalFunSuite to spinal lib.
fix #797
closed 2022-07-23 by Dolu1990