SpinalHDL/SpinalHDL #805: Bug? Found `NullPointerException` in `PhaseVerilog.scala`

tanhongze, 2022-07-25:
When I replace my RAM with a black boxer, I get a `NullPointerException` in `PhaseVerilog.scala`.
```
case bb: BlackBox if bb != null => {
       val str = bb.impl.getVerilog()
    ...
}
```
Here, `bb` is not null but `bb.impl` is null. How to fix it?
And is there any way to fix it in my code before next Spinal release?

tanhongze, 2022-07-31:
@Dolu1990 ?

Dolu1990, 2022-08-02:
Hi ^^

Sorry, i was gone in holiday, AFK :D

Yes you are right, it was recently fixed : 
https://github.com/SpinalHDL/SpinalHDL/blob/1514ec64d47b0efe7c73ea69d846c0b51330924f/core/src/main/scala/spinal/core/internals/PhaseVerilog.scala#L106

and released, in SpinalHDL 1.7.2

closed 2022-08-02 by Dolu1990